sailoog/openplotter #10: Kplex config compatibility

joabakk, 2015-09-13:
Would it be an idea to be able to show and edit the kplex config from the menu, possibly switching between the menu generated conf and a detected change? When doing manual changes in conf, these are just lost currently when the nmea menu is opened

sailoog, 2015-09-14:
OpenPlotter aims to be easy for users and easy for developers :) so we did not intend to create a complete kplex GUI and we used only that features of kplex that we needed. But you are right, it did not occur to me that we could allow to edit the kplex config file manually (adding a button to open it). The problem is, as you say, to maintain these manual settings. Now you can edit manually the kplex config file but when you apply any change by the openploteer interface, these changes are overwritten. We will need to process the config file in another way than actual to compare changes. Let me study the code but I think is a very good idea and deserves to be in the next release. Thanks!

joabakk, 2015-09-18:
Hi again,

It appears that my old working kplex.conf is not working, as the try fails and openplotter.py won´t start. The config file is something like this:

[global]
#failover=GP***:0:nx:30:mux:60:ais
checksum=yes

#Arduino kompass
[serial]
#filename=/dev/ttyUSB0
filename=/dev/ttyACM0
name=compass
#srctag=yes
#timestamp=s
direction=in
baud=38400
#baud=4800

#Multiplexer
[serial]
filename=/dev/ttyUSB2
name=mux
srctag=yes
#timestamp=s
direction=both
baud=4800
optional=yes

#Nexus server
[serial]
filename=/dev/ttyUSB0
name=nx
srctag=yes
#timestamp=s
direction=both
baud=4800
optional=yes

#AIS
[serial]
filename=/dev/ttyUSB1
name=ais
srctag=yes
#timestamp=s
direction=both
baud=38400
optional=yes

#AISrx
[serial]
filename=/tmp/aisrx
name=aisrx
srctag=yes
direction=in
baud=38400
optional=yes

#AIS rx #if sent from different rpi over tcp
#[bcast]
#name=aisrx
#port=10111

[tcp]
mode=server
port=10110
direction=both

#[bcast]
#device=eth0
#address=255.255.255.255
#port=55554

[file]
name=log
filename=/home/pi/kplexlogs/nmea.log
perm=775
direction=out
srctag=input
timestamp=s
append=yes
eol=n

[pty]
name=freeboard
mode=master
filename=/dev/pty22
baud=38400
group=dialout
perm=0660

[pty]
name=pynmea
mode=master
filename=/dev/pty23
baud=38400
ofilter=+**ROT:+**RPM:+**HDM:+**HDT:+**VPW:+**MWV:+**VHW:+**VWR:+**GGA:-all
srctag=input
timestamp=s
group=dialout
perm=0660
#direction=both

sailoog, 2015-09-18:
Never will. OpenPlotter is not ready to accept manually edited conf files, just settings created by itself. It is not a trivial change because I have to change completely how the file is processed. I will do ASAP.

sailoog, 2015-10-07:
kplex developer is working on a new way to interface kplex so I think I had better wait for it.

sailoog, 2015-11-02:
Done!

I have found a way without having to rewrite the script. At the end of the kplex configuration file there will be a tag "###Manual settings", All you put there will be respected by OpenPlotter GUI :)

This is an example:
```
# For advanced manual configuration, please visit: http://www.stripydog.com/kplex/configuration.html
# Please do not modify defaults nor OpenPlotter GUI settings.
# Add manual settings at the end of the document.

###defaults

[udp]
name=system_input
direction=in
optional=yes
port=10110

[tcp]
name=system_output
direction=out
mode=server
port=10110

###end of defaults

###OpenPlotter GUI settings

[serial]
name=gps
direction=in
optional=yes
filename=/dev/ttyACM0
baud=9600

###end of OpenPlotter GUI settings

###Manual settings

[file]
name=log
filename=/home/pi/kplexlogs/nmea.log
perm=775
direction=out
srctag=input
timestamp=s
append=yes
eol=n
```

closed 2015-11-02 by sailoog

joabakk, 2015-11-02:
Wonderful, thanks a lot