sailoog/openplotter #9: Rate of turn calculation and nmea generation from imu sensor data

joabakk, 2015-09-13:
What would it take to include rot calculation in the sensor tab? A calculation in sensors.py and a setting in config, anything more? I am askimg so that I can contribute

sailoog, 2015-09-14:
There is already a NMEA generation rate field

[image]

joabakk, 2015-09-14:
Thanks, it generates HDG just fine. I was talking about ROT (rotation speed). It is often used as a necessary AIS input and for some sailing apps (including my own python polar diagram saver).

sailoog, 2015-09-15:
ups! sorry I did not know that sentence and thought it was a typing mistake (putrefaction calculation? oh my god) sorry English is not my native language :)

Sounds great!. I have see that you have already implemented that sentence in pynmea2.

I suppose that you need magnetic heading and time values to calculate it and since the IMU sensor is not the only possible source of heading I think the best place to do that is calculate.py.

You will need:
create interface elements in openplotter.py and add some function to calculate block from line 1120
add item in openplotter.conf ([STARTUP] nmea_rot = 1)
function in calculate.py after #GENERATE NMEA label

I can do openplotter.py changes if you want.

Thanks!!

joabakk, 2015-09-18:
I´ve proposed openplotter.conf and calculate.py changes in the last pull requests #11 and #12 . The calculation is still missing some interface to the timestamps, and the actual NMEA sentence generation.

sailoog, 2015-09-18:
Ok, I will create a branch to test. I have noticed a possible error in your code and commented.

sailoog, 2015-09-18:
Done. And I have created the interface elements in Calculate tab and the functions in the main scrip openplotter.py
On my branch ROT there are your calculate and conf changes + my last interface changes. You can apply my changes to your fork or maybe better make a fresh fork again.

joabakk, 2015-09-18:
Wow thanks!

joabakk, 2015-09-22:
I just sent a proposal. It is working, but there may be a need for a dampening setting, since the values are very high

sailoog, 2015-09-22:
It seems that all is ok, if it works, as you say it will be a matter of fine calculation.
I have modified startup.py file to start rot generation at startup if selected.
Good work!

joabakk, 2015-09-24:
Minor change in latest pr. Case closed. Thanks a lot

closed 2015-09-24 by joabakk

sailoog, 2015-09-24:
merged! I will test in deep when I finish the implementation of heel in the same branch. Thanks to you!

sailoog, 2015-09-25:
I think it is working right. Values are high because I suppose they are degrees per minute, aren't they?

I have deleted the variable "rot" initialization because it was not needed and rounded the final value to just 1 decimal. Maybe it would be good rounding to no decimal as well.

joabakk, 2015-09-25:
They seem to be correct. I had set the threshold in my other app to be "stable course" if ROT was less than 5 deg/m, but I see the value is 6 deg/m if 0.1 deg difference in 1 sec which makes sense

I look forward to testing the updates on ROT and heel. Great work on combining all nice things into one package!