sailoog/openplotter #15: Additional NMEA sentences

gvaughan67, 2015-09-18:
Could we create a heel angle sentence from the IMU? I've seen the XDR sentence used for that. There are $700 black box devices to do that in the marine catalog. :) 

Also, the Air temperature sentence created by OpenPlotter from the barometer isn't the one that OpenCPN Instruments plugin is looking for.

I am keen to add voltage monitoring (a divider circuit and an analog/digital converter) and water temperature sensors and sentences.

sailoog, 2015-09-18:
I am afraid it is not possible to create wheel angle with the current IMU. The IMU sensor measures the angle of the boat from North, and you would need another sensor to measure the angle of the wheel from the centerline of the boat. Maybe an IMU mounted on the rudder axis...?

Yes, you are right, I noticed that a few days ago. Do you know what sentence is? Actually, according to the NMEA entity, the sentence OpenPlotter uses for pressure and temperature is obsolete (MDA) and the substitute is just XDR.

Great! both features are in the TODO list. I had thought to do engine and water temperature with high temperature and waterproof digital sensors (DS18B20). Any idea to make the voltage monitoring?

gvaughan67, 2015-09-18:
Not “wheel.” “Heel.”

En Español: Escora

http://icdept.cgaux.org/pdf_files/English-Spanish-Glossary-Nautical-Terms.pdf

Both of our languages have such a rich and distinct vocabulary for old sailing terms that mean nothing to people who don’t sail, so are hard to translate.  ☺

For voltage monitoring I am going to try two things. First an Arduino with a divider circuit. Then an analog to digital converter circuit with a divider circuit with the RPI. I’ve ordered the parts.

http://www.lediouris.net/RaspberryPI/ADC/monitor-12v.html

joabakk, 2015-09-18:
I think heel angle would also be XDR, as far as I recall. The freeimu code sends out heel and pitch on XDR sentences. Now the advantage of openplotter is that the code can be aligned on both sender and receiver for the code used. As long as we know what opencpn requires from heel sentence

sailoog, 2015-09-21:
ops! sorry :cold_sweat:

So, you are right hell calculation it will be great too. What a useful sentence...

As far as I know...

```
$--XDR,a,x.x,a,c--c, ..... *hh<CR><LF>

a= Transducer Type (P: pressure, C: temperature, H: humidity, U: voltage, I: Intensity)

x.x= Measurement Data

a= Units of measurement (B: hpa, C: celsious, V: volts, A: amp)

c--c= Name of transducer

*hh= Checksum
```
I think picht, roll, yaw are:
```
Transducer Type (A: angular displacement)
Units of measurement (D: degrees)
```
Can you confirm joabakk?

Look at this voltage measurement without calibration:
http://raspi.tv/2013/controlled-shutdown-duration-test-of-pi-model-a-with-2-cell-lipo

We are working on a waterproof case so reduced size will be great for that circuits.

Thanks!

joabakk, 2015-09-21:
That sounds about right, see the code and comments here https://github.com/rob42/FreeIMU-20121106_1323/blob/master/FreeBoardIMU/FreeBoardIMU.ino

As for the lipo link, I think we should combine a hat with gracefullty power down pi, inputs from engine acc line (to detect engine running), battery sensing, analogue inputs from engine temp etc...
12v Use: car, boat, camper etc
RTC, circuit from wifinmea
hat design based on rpi-proto-hat (github)
And the imu should ideally be away from the rpi and instrument panel, so a harness for this cable

I have started this and will share,

sailoog, 2015-09-22:
ok, copying here to remind
```
$YXXDR,<1>, <2>, <3>, <4>,<5>, <6>, <7>, <8>,<9>, <10>,<11>,<12>,<13>,<14>,<15>,<16>*hh<CR><LF> 
eg $YXXDR,A,5.2,D,PTCH,A,7.4,D,ROLL,,,,,,,,*hh<CR><LF>
The fields in the B version of the XDR sentence are as follows:
<1>A = angular displacement
<2>Pitch: oscillation of vessel about its latitudinal axis. Bow moving up ispositive. Value reported to the nearest 0.1 degree.
<3>D = degrees
<4>PTCH (ID indicating pitch of vessel)
<5>A = angular displacement
<6>Roll: oscillation of vessel about its longitudinal axis. Roll to the starboard is positive. Value reported to the nearest 0.1 degree.
<7>D = degrees
<8>ROLL (ID indicating roll of vessel)
<9>+Not used
```

yessss, I started a Hat design too :) 

Do you want to open a new issue where we can collect all the possible options?

joabakk, 2015-09-22:
Sure, let's do that

joabakk, 2015-09-23:
it appears from https://github.com/richards-tech/RTIMULib/blob/master/Linux/python/tests/Fusion.py that roll (heeling angle or list) is (math.degrees(fusionPose[0]) and pitch (trim) is (math.degrees(fusionPose[1]). It should then only be to add this in settings and openplotter.py and copy and adapt the nmea sentence generation in sensors.py to suit. I don't have time just now, but if you prefer, I can suggest sensors.py changes

sailoog, 2015-09-23:
Yes you are right but let me study that because first I have to migrate the obsolete MDA sentence (pressure, temperature) to the recommended XDR. 

XDR generated sentence should contain all data (press, temp, heel, the future humidity...) in a single sentence to avoid saturating the NMEA stream.

joabakk, 2015-09-24:
OK good,
It might be an idea to be able to "reset" the y,z axis and correct the heading after installation on board to show realistic pitch, yaw and roll values

gvaughan67, 2015-09-24:
Another "reset" function that would be useful for some is an adjustment to apparent wind direction: -x degrees or +x degrees. There is a hardware reset function on my nmea wind instrument, but it requires pointing directly into the wind and shorting two wires at the right moment.

sailoog, 2015-09-25:
Done!

ROT and Heel implemented on ROT branch. Air temperature and pressure migrated to XDR sentence and working like a charm.

Please test (with IMU sensor if possible) and I will merge it to the master branch.

Thanks guys!

sailoog, 2015-09-25:
About "reset" functions....

I think it is not necessary to do that with compass/yaw value because the reference always will be North and a good calibration will work.

It is different with heel/roll value because the reference will be the boat and it may be necessary this "calibration" after installation. Same for apparent wind direction. Are large deviations a common issue?

joabakk, 2015-09-25:
We'll have to see next summer when it's all working and we have installed the sensors on board? What I've seen is that some heading sensors are able to rotate after mounting to align the true north to the installed heading on board, 
[image]
or that you can rotate the mounts somewhat to compensate
[image]

This can be overcome by setting a true value/correction. The alternative is some kind of swivel connection of the sensor connection.

From my basement testing, the heel value (on the calibration page) seems to be misaligned when the imu is horizontal. This would equate to a constant list which could be corrected in the settings.

joabakk, 2015-09-26:
The heel angle and Rot are working just fine. I don't know any apps or pprograms that use heel angle, or any other XDR data, so the test is based only on imu testing in the view nmea page.

sailoog, 2015-09-29:
I have merged ROT and Heel to master branch. Let's test it in real life. About "reset" functions, it would be better to open a new issue if necessary so I can close this.

closed 2015-09-29 by sailoog