sailoog/openplotter #18: OpenPlotter Hat - features proposal

sailoog, 2015-09-22:
Raspberry Hats are really small and we would have limited space to make all of this but we could make more than one board. Here we go...

#### I2C screw connector (4 pins).
IMU, barometer, humidity sensor and others, have to be connected in parallel to port I2C

#### 2 Screw connectors for 2 customizable switches (2 pins x2 + 2 resistor)
Shutdown, start services...

#### 1 Screw connector for multiple waterproof/high temperature digital sensors (3 pins + resistor)
Engine and water temperature, etc. DS18B20 sensors.

#### Voltage monitoring
http://www.lediouris.net/RaspberryPI/ADC/monitor-12v.html
http://raspi.tv/2013/controlled-shutdown-duration-test-of-pi-model-a-with-2-cell-lipo
arduino?

#### Seatalk
yappelectronics boards?

#### RS232
http://codeandlife.com/2012/07/01/raspberry-pi-serial-console-with-max3232cpe/

#### CAN bus
http://www.industrialberry.com/canberry-v-2-0/

#### 1 Screw connector for IR motion sensor (3 pins)

Notes:
Most of sensors or stuff to connect could share GND, Vcc?

joabakk, 2015-09-22:
Good start. This will very fast become a wish list where we need to prioritize, but here we go

-RTC, because they are small and we would not lose time even with no network and no gps

-The circuit to gracefully shut down the pi from a switch (need one always on and one switched, a relay and some caps to keep the relay "charged" from the signal until completely shut down. I use 45 secs on my prototype, works fine. This was my start because i run headless and I need to keep the pi sound and safe.

RS232 is not necessary for my part, I have plenty adapters from USB.

sailoog, 2015-09-23:
if that's okay I will strikethrough the discarded features adding the new ones and highlighting the accepted ones.

I do not like RS232 either, sure there is a better way to use RX and TX GPIO pins.

RTC is ok but do you think it is essential?

Your circuit to shut down seems a little bit complicated, how about this?
(Safe Shutdown Switch) http://www.raspberry-pi-geek.com/Archive/2013/01/Adding-an-On-Off-switch-to-your-Raspberry-Pi

I think we should mark "I2C screw connector (4 pins)" as essential.

joabakk, 2015-09-23:
Good idea. I added the RTC so that I had a prioritized time from NTP>GPS>RTC. It is not super important for me

The circuit I use is based on http://electronics.stackexchange.com/questions/61877/shutdown-controller-for-raspberry-pi-in-a-car
combined with http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=55410 and 
the code from: https://www.pi-supply.com/pi-supply-switch-v1-1-code-examples/
The advantage is that the switched power is directly from my instruments switch, so there is no extra switch to remember when I leave the boat. A separate switch is much less complicated of course

I agree, i2c is essential

nicolas-j, 2015-09-24:
Hi all,
About the power feature, I have been using this a bit : https://pi.gate.ac.uk/pages/mopi.html,
nice feature is the use of several power sources.
And this one, wich provides a clean shutdown: https://www.pi-supply.com/.
Might be a good reference.

joabakk, 2015-09-26:
Canbus, is this a replacement for the actisense n2k to nmea 0183 usb? If so, the yes please!

sailoog, 2015-09-29:
Not exactly, it is a replacement for the actisense n2k to raw serial data, not nmea 0183. Signal K it is charge of translating from raw serial data. This have to be confirmed because NMEA 2000 has their own CAN specification but I know that there are people getting NMEA 2000 from Beaglebone CAN bus.

joabakk, 2015-10-02:
Ok let's figure that out some time. If that is possible, then yes that would be a valuable addition.

Another valuable thing that I just remembered I have now is one or two 12v inputs with voltage dividers. I use one to see if the engine is running or not (acc output from ignition). We could also have one analogue input to sense battery voltage and possibly another for battery voltage after a shunt to calculate current consumption from the difference (I have that on a separate instrument, but that is standalone. 

This is my powerHat design, with shutdown function, RTC and voltage divider input:
[image]

sailoog, 2015-10-07:
Your design looks great! I think we need to set some hardware API or "rules" to avoid Pins conflict so every OpenPlotter Hat are compatible. I mean...
I am finishing of implementing  the most of the OpenPlotter core features (sensors, switches, etc.). These features need some specific GPIO pins, some screw connectors and some resistances to work. It would be great that every OpenPlotter Hat (as yours) incorporates these components and connectors and fits the pins requirements to keep compatibility. If that is not possible due to problems of space, it should fit pins requirements at least to allow connecting a plain "connections Hat" with just connectors and components as well.

What do you think?

joabakk, 2015-10-09:
Well I'm using IO4 for the switching sensor,  and IO21 to sense if the engine is running or not. I use yet another pin (IO20) to check if the hat is "mounted" or not. These are far from proposals for standards. I think the idea of using "standards" is ok, but it will be difficult to maintain. If we agree to a design somewhere, we can all add to the design and use the unused pins. We all have different needs and I don't see us mixing eachother's hats, but some borrowing of principles may exist. I have left room for some custom pins to be used breadboard style, but eventually we will run out of space in an enclosure. My boat currently has my design, but many components like the relay is outside the box (standard rPi enclosure)

muckleyj, 2015-10-09:
What if the hat and software just provided a  more flexible and universal
input?  I'm thinking a high impedance volt meter on every input and a
regulated voltage supply.  With a proper user interface every input could
sense whatever you want it to.  Battery voltage?  OK.  Shunt voltage
(current)?  Sure.  Any temperature sender?  Yep.  Pressure sender?  Yes.
Fuel tank float sender?  Can do.  Propane tank sender?  Covered.

The software would just need to provide a means of calibration, function,
and distribution (signal K).

An extension of this as a universal input platform would be a universal
output platform too.  I can't think of too many uses for a variable output
so it could be constrained to simple On/Off commands driving solid state
relays to control buzzers, lights, appliances, etc.

I know it's more to add to the wish list but what if the HAT included a
small battery and a shutdown script so that during a short loss of power
the PI would stay on but when you intentionally turn the instrument panel
off, the PI performs a proper and safe shutdown?

Josh Muckley
S/V Sea Hawk
Solomons, MD

sailoog, 2015-10-12:
I was working on a more "hardware" solution in order to make it safer, I mean... the new version will have a new tab *Switches* where one will be able to link actions to switches. There are special switches that could be connected as well like IR motion sensor, water level sensor and magnetic switches (to detect doors opening). These planned switches will be able to trigger alarms and remote monitoring actions in the near future. 

GPIO pins need pull-down-up resistors to work properly and that was the reason I was trying to suggest to include resistors and connectors to hats. But... you are right that is not flexible so I have provided a more "software" solution enabling the internal raspberry pull-down-up resistors.

[image]

You can choose between pins 22, 23, 24, 25, pull down or pull up, an action for each state and a command if you select command action. In the circuit you will not need resistors, just connectors. The only reserved pin will be IO4 because DS18B20 sensors require that one.

So... the suggestion would be that future hats do not use pin 4 and add connectors for 4, 22, 23, 24, 25, Vcc (3,3v), Vcc (5v),GND, SDA, SCL in order to allow users to take advantage of all openplotter and hat features.

I could create an eagle file with a "template" for this.

**joabakk**

Could you adapt your hat to this? 
Since you are using eeprom you could use it to detect if hat is mounted by I2C.

**muckleyj**

There are some hats with that "safe shutdown" feature. It would be great to install super-capacitors instead battery.

joabakk, 2015-10-12:
Sure, this sounds good. I had planned to use the EEPROM to identify the hat, but never got around to learning how and to enable the code portion. But I'm flexible.  

My hat gets around the battery or super-cap issue because the cap only stores enough oomph to charge the transistor until the pi is ready to shut down. Just an idea.

NAHANNIV, 2015-11-05:
I know the Pi is very popular, but it is not the only platform.

What about a separate board instead of a Pi specific HAT ?

Perhaps connected by a short I2C lead.

sailoog, 2015-11-09:
Yes, it would be the best. OpenPlotter goal was to be multi-platform but has become more hardware dependent than we wanted. Not only I2C port is used, I/O pins are needed and every platform has different interfaces.

hwater, 2015-12-01:
please release your "Switch" Interface for Pi as soon as possible. Maybe other Platforms are faster, but the new mini-pi prize is unbeatable. An litte Quirk: Now my I2C Bus is only populated with an BMP085 - this is only sometimes detected, but mostly your Program detects it as an MS5611?. 
I will try to connect some DS18B20 Tempsensors to GPIO4 in the next Days. 

Many Thanks for all your good Work & many Greetings from the Danube (km2133) 
holger - donautics.stwst,at

muckleyj, 2015-12-01:
Hey yall,

Just wanted to let you know I've purchased and tested the UPS PIco from
www.Pimodules.Com.  It provides a configurable file safe shutdown feature
with a backup battery.  This would allow you to power the PI from a remote
power switch, say on your power distribution panel.

Just so everyone knows this particular HAT uses GPIO 22 & 27.

Warm regards,
Josh Muckley

hwater, 2015-12-02:
Hello SailLOOG - just updated to the latest git version - all ist running fine now with the I2C-Bus.

sailoog, 2015-12-03:
Here we go :)

This is a beta version for evaluation and development purposes only.

http://www.sailoog.com/en/blog/download-openplotter-rpi-v060-beta3

NAHANNIV, 2016-05-22:
Thinking about this a bit more as I am trying to figure out how to install this on my boat.

I think that a HAT design may not be the best Idea, since it seems that my Rpi3 will require a (large) heat-sink. Running a RADAR display in OpenCPN raises the SOC temp above 80deg.C even with ambient temperatures of 17deg.C. I think this could be a problem when ambient temperatures get up to 30deg.C. The problem will also be made worse when the board is installed in a waterproof (or drip-proof) case.

What I am working on is a heat-sink larger than the size of the Pi with large fins on the back to which the Pi gets screwed down with the SOC side toward the heat sink (thermally connected by a copper block).

The Pi and cables will be covered by a three sided plastic, gasketed cover also screwed to the heatsink. The whole assembly is intended to be mounted vertically with the open side of the cover facing down (drip-proof). The HDMI cable needed a 90deg. adapter.

My point relating to the HAT is that to fit in with the case/heatsink design outlined above, I think a board mounted beside the Pi would be better. That would allow it to use the heat sink as well.

Some things I would like to see on that companion board are: Autopilot motor driver circuit (attached to heat sink), wide input power supply circuit (I don't like the USB power cable), CAN/NMEA2000 interface, connectors for I2C, 1W, etc.. 

Cheers,
JM.

sailoog, 2016-05-23:
You may be right. We are thinking of something to connect to the raspberry but perhaps we should start thinking of connecting the raspberry to something.

closed 2017-08-27 by sailoog