sailoog/openplotter #21: Additional software for Openplotter RPI image

gvaughan67, 2015-09-24:
These are just ideas, and possibly bad ones. This is a low powered CPU and there are limits to what we should ask it to do.

1. Control of navigation lights and other DC circuits using a relay board. This would be a simple web page with buttons executing server side scripts to turn lights on and off, accessed usually from a mobile device in the cockpit. One could get fancy and use light sensors to turn anchor light off in daytime, and running lights on when underway at night. http://www.banggood.com/5V-8-Channel-Relay-Module-Board-For-Arduino-PIC-AVR-DSP-ARM-p-74110.html?currency=USD&createTmp=1&utm_source=google&utm_medium=shopping&utm_content=saul&utm_campaign=Electronic-xie-us&gclid=Cj0KEQjwvo6wBRCG3Zv92ZSLlIYBEiQA5PLVAjqgzuXoLSXVeBP5dN-Ef8HWF4P3tm9mozZLRdfbUngaAq2N8P8HAQ

2. Weather station software, analyzing apparent wind (relative to NORTH not course) and barometer history, and giving forecasts from same.

3. MPD (Music Player Daemon) music server. Allows a robust interface for playing back music, using mobile apps and web clients. 

For the flexibility of using wireless mobile devices to move about the boat, I generally have a preference for responsive web applications over linux desktop applications, although it is hard to deny the usefulness of OpenCPN and a real gui for configuring things.

joabakk, 2015-09-25:
I agree that there are many things we can do with op as a starting point. But it's a small computer and not all will need all things. Most of this is for the user to install on top. But some additions could be presented as an "app store" to add functions.
1. this is fancy and could be nice to have, but not necessary for me at least
2. This sounds complicated. I would rather have route planning based on polar diagrams (for sailing), but this should be possible from within openCPN
3. Not specific for a boat rPi, should be fairly easy to add if you find a turorial

I have some scripts to create dynamic polar charts based on highcharts. Example here. This requires mysql and highcharts
[image]

joabakk, 2015-09-26:
How about CANboat?  Combined with sensors eg. On the engine, it should be possible to send n2k only data to signal k. I'm new to signal K and n2k, but there are only so many types of data you can send over nmea0183. I have been wanting to monitor engine temp, rpm and tank levels, but there are no nmea0183 sentences. I believe this is the beauty og signal k

joabakk, 2015-10-01:
Also, I have been using freeboard on my previous image. This is a headless chart plotter and instrument gauge web interface. I see that there is work started to use signal k info as input instead of nmea0183.

sailoog, 2015-10-01:
gvaughan67:

1. That sounds good. To add these features we should decide either go on LAMP servers or node.js servers. If signal k will be essential in the future I think node.js should be the way.
2. OpenPlotter has already weather graphs but a complete weather software able to get NMEA data would be great. Do you know any open source and Raspbian compatible?
3. Here, I agree with joabakk.

joabakk:

Hey your polar software looks great. How could I test it?
I know freeboard but never tested. Do you mind writing some mini howto install it on raspbian?

joabakk, 2015-10-01:
The polar python scripts is my first experience with python, and I have to clean up the code and add more comments before I pass it on (at least remove the passwords to mysql). But I'm glad you like it. Basically it's a working hack at the moment.

Freeboard, the rpi installation is a script (but may have conflicts to your image) here:
https://github.com/rob42/freeboard-server/blob/master/install_rpi.sh

The signalk version looks less complete, but described here: https://github.com/SignalK/freeboard-sk

joabakk, 2015-10-02:
Hey,

I have chosen to add the polar scripts as is to https://github.com/joabakk/Dynamic-NMEA-polar-diagram 
You are welcome to suggest changes and improvements. It does not currently work with openplotter due to the kplex.conf restrictions. There are possibly better ways to get the timestamps (than tags on NMEA) and the NMEA stream.

gvaughan67, 2015-10-23:
I agree Polars are a better idea than anything I've proposed. :+1: 

It would be nice to see that merged with SignalK but I believe SignalK is mostly realtime data, not sure.

joabakk, 2015-10-27:
Thank you.

I agree that the polars themselves might need to be stored in a db, but for the communication back and forth, it is probably better to implement SK than to make custom NMEA sentences. Not least because the timestamps belong to the data, instead of using a tag on each sentence. I believe most are already in SK, but in my script, I'm writing to a json file with wind speed and direction, boat speed, polar boat speed, target boat speed and angle, performance, speed and cardinal direction of current, lat and lon. 

I can't see just now when I will have time to learn SK and change the scripts, but you are most welcome to fork and improve.

joabakk, 2016-02-08:
Would it be an idea to add to the gui some sort of "app store" which could add packages from pip, git, npm and so on? There are already quite a few packages to update, and it could be a way to upgrade these and ensure necessary changes and dependencies? Not sure it's even possible

jmduno, 2016-02-08:
Hi

Yes, like a ppa repository. As opencpn already does. 

Some other packages are already "debianised". They may have to be compiled for armhf. 

I am reading doc about it. 
It's not so obvious to do it good.

joabakk, 2016-02-09:
that could be an ultimate goal, "50 new openplotter apps today". 
For now I was more thinking to tick off openplotter, opencpn, kplex and or signal k etc. and press update. This could run all the scripts for the user.

zASSAz, 2017-02-27:
It will be fine to have anchor watch integrated. With alarm and SMS sending future.

closed 2017-08-27 by sailoog