sailoog/openplotter #26: Variabilizar el nombre del WiFi Access Point

ajros, 2015-09-29:
La idea es poder configurar diferentes puntos de acceso con con nombres diferentes: OpenPlotter1, Openplotter2,...etc. de forma de poder usar varas RPi en el mismo lugar físico. Se puede crear modificando la microSD, agregándole un archivo que contiene sólo un nombre. O en OpenPlotter, una vez ingresado realizar la configuración. Si se elige la opción primera, automáticamente pasa a OpenPlotter y ya no es necesario volver a modificar la microSD.

sailoog, 2015-09-29:
Recibido ajros. Lo pongo lo siguiente en la lista. Gracias!

sailoog, 2015-10-01:
Hecho!
Añadido campo SSID para diferenciar redes y procedimiento para arrancar el sistema por escrotorio remoto sin necesidad de monitor. Gracias!

---------------------------------------------------------------------------
Done!
Added SSID field to WiFi settings to differentiate networks in the same place and procedure to set screen resolution and access point settings in order to start the system without monitor (headless)

closed 2015-10-01 by sailoog