sailoog/openplotter #27: Wifi access point password

joabakk, 2015-09-30:
Hi again,

One thing I noticed with the OP image, is that when I enter a password for the wifi ap and enable it, I can acess the ap, but I'm never asked for any password on the client. My password is more than 8 characters. This is beyond me, but I've used the adafruit guide (https://learn.adafruit.com/setting-up-a-raspberry-pi-as-a-wifi-access-point/overview) before, which suggests this:

interface=wlan0
driver=rtl871xdrv
ssid=Pi_AP
hw_mode=g
channel=6
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=Raspberry
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

sailoog, 2015-09-30:
This has never happened to me. Are you sure your client is not remembering the password?

joabakk, 2015-09-30:
Well wouldn't you know, now it works. Sorry I must have done something strange last time. Never mind!

Joachim Bakke

This e-mail and any attachment is strictly confidential and is intended only for use by the addressee unless otherwise indicated.

closed 2015-10-01 by sailoog