sailoog/openplotter #28: IMU settings to reset orientation

joabakk, 2015-10-01:
as mentioned in the closed issue, it would be good to be able to align the orientation after the imu has been mounted. We could take a gps fix on the bow, walk a few hundred meters and take a new fix to get an exact boat heading, walk back and add or subtract the difference in heading between gps and compass. Also, the roll /heel may be corrected when the boat is level.

sailoog, 2015-10-07:
I have made some improvements and debug in IMU code which I would want to test in the next release. If we get good feedbacks I will implement this "reset" feature you suggest.

joabakk, 2015-10-09:
I have tried to pull your code and test it, but I can no longer open openplotter.py from the start screen. No Idea what happened

sailoog, 2015-10-10:
Install https://github.com/timofurrer/w1thermsensor

joabakk, 2015-10-11:
Thanks, that worked. But maybe it's n idea to test the presence of this library in the code, instead of crashing? This means it's no longer a git clone of this repo.

closed 2016-01-25 by joabakk