sailoog/openplotter #39: Missing module

joabakk, 2016-01-25:
Hi again,

I have updated to the last master repo and I get an error message when I try to launch openplotter.py: 
Traceback (most recent call last):
  File "openplotter.py", line 22, in <module>
    from classes.actions import Actions
  File "/home/pi/.config/openplotter/classes/actions.py", line 19, in <module>
    from classes.twitterbot import TwitterBot
  File "/home/pi/.config/openplotter/classes/twitterbot.py", line 18, in <module>
    from twython import Twython
ImportError: No module named twython

Where is this library, or do I have to manually add it somewhere?

sailoog, 2016-01-25:
Hi,

I am about to publish a new release witj lots of improvements. You should download the beta image version wchich comes with all the required modules and update openplotter to the last master repo:
http://www.sailoog.com/en/blog/download-openplotter-rpi-v060-beta3

Or wait a week to download the new stable release :)

closed 2016-01-25 by sailoog

sailoog, 2016-01-25:
Anyway, search on my repos. I am going to maintain updates clones from all the needed modules
```
cd
git clone https://github.com/sailoog/twython.git
cd twython
sudo python setup.py install
```

joabakk, 2016-01-25:
I pip-ed it, and now it works. Thanks a lot.