sailoog/openplotter #43: Multipe Acces VNC

hwater, 2016-02-24:
Add Multple Access & Use Password to VNC in File startup.py

#	subprocess.Popen(['x11vnc', '-forever'])     
	subprocess.Popen(['x11vnc', '-usepw', '-forever', '-shared'])

sailoog, 2016-03-02:
I agree with -shared but we should implement some way to set the password, any idea?

hwater, 2016-03-03:
About the password:

i would not go too far into the system - some Dokumentation should be enough.
Most People only set it one Time….

Keep on with your great Work

Holger Watermann

> 
> I agree with -shared but we should implement some way to set the password, any idea?

—  
Holger Watermann
Administrator
Information Management

JOHANNES KEPLER
UNIVERSITY LINZ
Altenberger Strasse 69
Hochschulfond Building, HF9903
4040 Linz, Austria
P +43 732 2468 3927
holger.watermann@jku.at <mailto:holger.watermann@jku.at>
www.jku.at/im <http://www.jku.at/im>

e-sailing, 2016-03-15:
I use xrdp. I can open more desktops on different clients with the same username.
I don't know if it is the better solution.

Martin

sailoog, 2016-03-16:
It's better for me too but  VNC is most popular

hwater, 2016-05-03:
Hello sailoog

please add the two optional checkboxes for VNC enabler in the Startup Page.
x "enable Multiuser" ('-shared')
x "use Password" ('-usepw')

thx (i have to edit them in after every clone for using "ScreenSharing" in MacOS X) 
or make them into one Checkbox called "MacOS X compatible"

sailoog, 2016-05-04:
I was delaying this suggestion because I didn't know how to deal with the password but I have an idea now, let me try...

sailoog, 2016-05-04:
Done. It will be always shared and password optional. Test and confirm please.

hwater, 2016-05-06:
Well done, works here. One Quirk, if you check the Passwd-Box for second Time, it should give you an Requester "Do you really want to change your VNC-Passwd ?" with "Cancel" & "Okay" before opening the Terminal.

Oherwise well done!

sailoog, 2016-05-06:
Done, well, something similar :) the requester will run always. Thanks.

closed 2016-05-06 by sailoog