sailoog/openplotter #44: Units in graph

richardmbell, 2016-02-26:
For those of us that use other units, could we support alternate units in the Therm/Bar/Hyg graph with something like the below?  Maybe have a user interface in OpenPlotter to select units and it could save the settings in a conf file.

pressureUnits = "mmHg"
pressureConvMult = 0.02953

temperatureUnits = "F" 
temperatureConvMult = 1.8
temperatureConvAdd = 32

pressure.append(round(float(log_list[i][1]) * pressureConvMult,1))
temperature.append(round(float(log_list[i][2])* temperatureConvMult + temperatureConvAdd,1))

desolina-nz, 2016-02-26:
It might also be good to record the temp/press/hum over say a 10min average. I think the csv also only holds 24hrs of data before over-writing. I've been increasing this to a week as it helps with tracking the weather using the pressure. 

It would also be great to be able to log more information like true wind speed, depth, location, etc. without having to use opencpn. I guess this would mean rearranging some of the scripts as only temp/press/hum are created in the i2c.py script where the current weather log is written.

sailoog, 2016-03-02:
@richardmbell You are right. Noted. I will study the best way to do this.

@desolina-nz in folder classes there is a script named datastream. This script can manage all the data flowing through the system (NMEA, switches and outputs and signalk in the future). What you say is interesting, there is a project of creating a web site to get data from openplotter installatons (openplotter meets IoT :) ) and some pre-buffer will be necessary. Maybe storing data, and sending averages...
Do not make nothing with datastream file until v0.7.0 stable is released because it will change a lot.

closed 2017-08-27 by sailoog