sailoog/openplotter #45: Inspector: Failed to connect with localhost:10110. Error: 111

oopskoule, 2016-03-01:
Hi,
When opening the  inspector, I get the error
"Failed to connect with localhost:10110. Error: 111, trying to connect..."

This occurs with Version 0.6.0 as well as with version 0.7.0beta.

Any idea?
Thanks a lot,
Peter

sailoog, 2016-03-02:
Hi Peter,

Do you have any input defined in NMEA 0183 tab?
Are you getting some error? see Reporting bugs here:
http://www.sailoog.com/en/blog/test-openplotter-v070beta

oopskoule, 2016-03-04:
OK, I got it fixed:
GPS mouse, and GPRS modem and other items were connected via a exteral USB hub.
At least connecting both directly to raspberry fixed the problem.
(keyboard, mouse, DVB-T-Receiver are still connected via USB-hub).

Now its working like a charme.
Great job!

closed 2016-03-04 by sailoog