sailoog/openplotter #48: Raspberry Pi 3

joabakk, 2016-03-09:
Is the image rpi3 compatible? I know it only just came out, but has anyone tried it? I tried the sd card from an earlier version, and it was dead in the rpi3

sailoog, 2016-03-10:
Sometimes yes and sometimes not :)
Some users are experimenting boot issues with openplotter v0.6.0 due to power feeding. It seems that you need provide the required volts. With the new official raspbian image that doesn't happen.
I am building the new openpotter image which will work on RPI 2 and 3. Give some days.

antti02210, 2016-03-11:
No success w rpi3 here either. Tried running the 0.6 image on rpi2 and upgrading the system, but run into problems with bootloader incompatibilities. Run out of tricks.
Great to know that a new image is brewing!

sailoog, 2016-03-11:
Read this please: http://www.cruisersforum.com/forums/f13/openplotter-project-152754.html#post2069418

thank you for your patience :)

antti02210, 2016-03-11:
> Lähetetty: ‎11.‎3.‎2016 17:25
> Vastaanottaja: "sailoog/openplotter" <openplotter@noreply.github.com>
> Kopio: "antti02210" <antti.jukarainen@iki.fi>
> Aihe: Re: [openplotter] Raspberry Pi 3 (#48)

Great, thank you for the link!
Antti

----- Alkuperäinen viesti -----

closed 2016-03-15 by sailoog