sailoog/openplotter #62: Installation docs

oudimou, 2016-04-23:
Hi,

I was about to start a fresh install of OpenPlotter on a UDOO board but I realized that **the installation doc is outdated** ("Obsolete, working on a new version!" written on top of doc since 28th January) and current effort seem mainly oriented towards RPI images.
Is there plans to update documentation so OpenPlotter can be installed on other boards, or even on a RPI without using provided base image ?

I'd be happy to help refreshing the installation docs (at least for UDOObuntu & UDOO Dual board) if you could provide me the necessary info, such as requirements (libs, python libs, etc.), what's obsolete in current doc, and what conf file are necessary to create/update.

sailoog, 2016-04-25:
Hi,
Yes, we are focussed on Raspberry in order to achieve goals ASAP and then work on migrations. I really appreciate your offer, resources are limited and maintaining the wiki was difficult because we invested all the time on development. I am about to release v0.8.0 stable and then I will know exactly all the dependencies and requirements. I could publish here or in the wiki all the dependencies but this could change in the next days until final release. I have set permissions to everyone to edit the wiki in case you want to make some work on it. What do you think? Could you wait some weeks or do you want I publish here some draft?

Thanks!!

plugje, 2016-04-25:
Hi Sailoog,

> I am about to release v0.8.0 stable

Can you give any clue when you expect this release? I've played with v0.6 and I'm about to do a clean install, which I prefer to do with the latest stable version.

greetz, Harry

oudimou, 2016-04-26:
@sailoog I can wait until 0.8.0 release :)
Please post dependencies once released (libs and softs), then I'll write down a step-by-step guide I used for my UDOO.
Im currently writing an Ansible playbook to automate OpenPlotter installation on my board (which might work on any board with an Ubuntu distro). So far I can install OpenCPN from PPA, and am on the way to implement zygrib building.

Regarding NMEA 2000, I intend to use https://github.com/ttlappalainen/NMEA2000, as the UDOO Dual also act as an arduino DUE. Will this work with OpenPlotter ?

sailoog, 2016-04-27:
@plugje 
Is difficult to say when v0.8.0 stable will be released but I hope no more than 2 weeks.

@oudimou 
That sounds great! I promise to update the wiki ASAP. That step-by-step guide will be really useful to port to other systems too.
I can not confirm if this NMEA 2000 solution could work. Perhaps. What is the output format? Have you considered [signal k](http://signalk.org/)?

oudimou, 2016-04-27:
@sailoog 
that NMEA2000 solution is supposed to act as a drop-in replacement for the Actisense NGT1 ... But I install it yet so I can't say more about it.
I consider signal k of course, but first I need to read from my instruments with NMEA2K to then expose collected data in signal k format and by wifi :)

sailoog, 2016-04-27:
We have hacked some low-cost car CAN-USB converters to work with openplotter in the same way.
They work pretty well so we are selling some of them soon :)

oudimou, 2016-04-27:
Hum. So I'll wait for this solution ... And switch my UDOO Dual for my PINE64 ^^

oudimou, 2016-06-24:
@sailoog Hey !
I saw 0.8.0 is out. Any news on the docs side ? Im about to buy a Raspberry and use provided image, so I won't bother about installation steps :)

oudimou, 2016-07-19:
@sailoog any news here ?..

sailoog, 2016-07-21:
Sorry @oudimou 
Lot of work and little time...
I will make some changes to the wiki today.

sailoog, 2016-07-21:
@oudimou
Done!
https://github.com/sailoog/openplotter/wiki

axelh, 2016-09-22:
oudimou, I just saw that you have the PINE64. Have you actually tried getting OpenPlotter running on it? I've got it running on RasPi B+, but it seems like it's way too slow. PINE64 seems like a better alternative to RasPi3 in my opinion, but probably lots of extra work regarding the port of all the software (just getting into all this stuff).

oudimou, 2017-02-22:
@axelh Heya !
Finally I used a RPI3 as I got a spare one from a colleague. So nothing to share on using OpenPlotter with the PINE64 :/

closed 2017-02-22 by oudimou