sailoog/openplotter #65: Cannot set static Ip for eth0 gets reset to dhcp address

antonical, 2016-04-26:
I cannot seem to set a static IP for the eth0 interface despite editing the connections. After each reboot the eth0 connection is recreated with dhcp.

Can you advise if this needs to be coded into a startup script somewhere.

Many thanks
Keep up the great work.

sailoog, 2016-04-27:
This is  the place to make your modifications: https://github.com/sailoog/openplotter/blob/master/wifi_server.py
line 101 if you are setting an access point and line 143 if not.

antonical, 2016-04-27:
Thank you,

I set the following in the else block. The interfaces file is written correctly but when the system boots I get the network is disconnected message. It says ethernet network device not managed. However the network is working on a dhcp connection and ignores all settings in the interfaces file, 

I am bit confused by this behaviour. Can you advise further.

This is the code I inserted:

else:
	data='# interfaces(5) file used by ifup(8) and ifdown(8)\nauto lo\niface lo inet loopback\n# The primary network interface\nauto eth0\niface eth0 inet static\naddress 192.168.0.7\nnetwork 192.168.0.0\nnetmask 255.255.255.0\nbroadcast 192.168.0.255\ngateway 192.168.0.254\ndns-nameservers 8.8.8.8 127.0.0.1 192.168.0.100'
	

This is the interfaces file that is written: 

# interfaces(5) file used by ifup(8) and ifdown(8)
auto lo
iface lo inet loopback
# The primary network interface
auto eth0
iface eth0 inet static
address 192.168.0.7
network 192.168.0.0
netmask 255.255.255.0
broadcast 192.168.0.255
gateway 192.168.0.254
dns-nameservers 8.8.8.8 127.0.0.1 192.168.0.100

All looks fine.

antonical, 2016-04-27:
Sorry needed to insert as code block:

`# interfaces(5) file used by ifup(8) and ifdown(8)
auto lo
iface lo inet loopback
# The primary network interface
auto eth0
iface eth0 inet static
address 192.168.0.7
network 192.168.0.0
netmask 255.255.255.0
broadcast 192.168.0.255
gateway 192.168.0.254
dns-nameservers 8.8.8.8 127.0.0.1 192.168.0.100`

LeaseOnLife, 2016-07-30:
I ran into the same issue. I only want a direct wired connection between PC and raspberry for nmea data, no internet routing. I added the static network settings for eth0 to /etc/dhcpcd.conf, added at the end of the file:
 interface eth0
 static ip_address=172.0.0.1/24

Seems to work now

sailoog, 2016-08-02:
We are rewriting the network part, we will also try to fix this. Thanks!

benjamingotke, 2017-06-26:
Hi
I have to connect my raspberry on a network of things with a ethernetcable. The eth0 should have following: address 192.168.1.5 and netmask 255.255.255.0. do I follow the same toturial to do this?

closed 2017-08-27 by sailoog