sailoog/openplotter #67: Cannot share mobile internet to WiFi hotspot

ryall, 2016-05-02:
I have an LTE (4G) dongle to connect to the internet via the mobile network, but the device does not appear in the "Internet connection to share" list when I set up my WiFi access point, only eth0 and wlan0, neither of which are the connection.

sailoog, 2016-05-02:
Just now I was checking and adding GSM dongles to share connections :)
I need to know how your device is detected. Connect to internet with your dongle, open a terminal and type ifconfig. You should get a list of your network devices: wlan0, eth0... and you should see something like ppp0. Can you confirm?
GSM dongles can be detected as ppp0, USB0 or even eth1, the last character could change.

sailoog, 2016-05-02:
Is your device in the list of supported SMS devices: http://wammu.eu/phones/ have you tested this OP feature?

sailoog, 2016-05-02:
fixed ppp0 devices in master branch

ryall, 2016-05-02:
Wow, that was quick! :)

Updating to latest to test now, will let you know. The device is a Huawei E3372 which doesn't seem to be listed in the SMS providers.

ryall, 2016-05-02:
Great news, it worked! Thank you for getting that in :)

One minor thing I noticed is that I had to be actually connected to the internet on the dongle before it would show up in the devices list. Just plugging it in and having it set up was not enough.

sailoog, 2016-05-03:
Yes, I know :( it seems to be a system policy that I don't know how to deal with. I have had some network manager crash when I have tried to make the mobile connection automatic. Let me know if you make some work on it. Thanks.

hwater, 2016-05-03:
> fixed ppp0 devices in master branch 

Please do the same for the usb0 device (LTE Modem MF831 from ZTE)

ryall, 2016-05-04:
@hwater Try this:

- Set up a new mobile network connection from the network connection manager.
- Connect to the internet on the device.
- Open the OpenPlotter app.
- You should see `ppp0` in the list of available devices.

As far as I'm aware, you shouldn't need to directly share usb0, just the network connection it creates.

hwater, 2016-05-04:
Hi ryall

This device creates an "usb0" in ifconfig, and not "ppp0" like other Devices.  This usb0 network connection gets it's own IP from the TELCO Provider. For now i handedit the openplotter.conf to share=usb0 and this shares the LTE Connection to all.

sailoog, 2016-05-04:
Ok no problem, it's an easy change but let me some time, busy now.

sailoog, 2016-05-04:
Done in master branch. Please test and confirm.

hwater, 2016-05-06:
Works fine here with this LTE Modem. Issure may be closed. Thanky

closed 2016-05-06 by ryall

snowboardergrun, 2017-02-04:
My huawei e5372 is showing with ifconfig like "wwan0" can you add it ?