sailoog/openplotter #80: ver 0.8 crash

fogns, 2016-05-23:
Hi,
Just testing out version 0.8.  It is hanging but I can still SSH in.

pi@openplotter:~ $ ps
  PID TTY          TIME CMD
 1736 pts/0    00:00:00 bash
 1792 pts/0    00:00:00 ps
pi@openplotter:~ $ python /home/pi/.config/openplotter/openplotter.py
Unable to access the X Display, is $DISPLAY set properly?
pi@openplotter:~ $

I rebuilt the SD card thinking I had a corrupted card before.  This happened when I plugged DS18B20 temperature sensor into the Pi3.  But this happened before with simply plugging the USB hub and rebooting.  

Thanks in advance, great concept!

James

fogns, 2016-05-23:
And it was working as I was able to set everything up.  Jc

sailoog, 2016-05-24:
I think you have to define some parameters before running a X app remotely.

If malfunctions happen when sensors or devices are plugged, could be a matter of wrong connections or low power supply? Test the new v0.8.0 please. Thanks for reporting.

fogns, 2016-05-24:
Hi 
I downloaded the latest 0.8 stable and it is now working fine.  I have all the sensors working now and have let it run over the last day or so.   The reason I wanted to post the detail was knew that you were working on getting a stable release together.  Didn't realize the time was when I was doing the 0.7Beta.. lol.
One thing I did find is not to use one of the new Amazon USB hubs... my keyboard and mouse become unusable.  Not recommended.  Great job on the development of OpenPlotter btw.  I develop node.js / javascript applications if I can be of any assistance.

closed 2016-05-25 by fogns

sailoog, 2016-05-31:
Of course you can ;) there is a lot of work to do in https://github.com/SignalK/instrumentpanel and my javascript is not good.