sailoog/openplotter #87: Starting OpenCPN

andyrbarrow, 2016-06-07:
Anyone else getting "Illegal Instruction" when trying to start OpenCPN on RPI2?

(When clicking the screen didn't produce a result, I tried the command line, hence the error message)

I tried purging OpenCPN and re-installing, no luck. 

Any thoughts on how I might troubleshoot it?

sailoog, 2016-06-07:
Could you paste here the error?
try opening opencpn with this command:
`opencpn -no_opengl`

andyrbarrow, 2016-06-07:
Unfortunately, there is absolutely no debug information in the command response.

pi@openplotter:~ $ opencpn -no_opengl
Illegal instruction

sailoog, 2016-06-07:
I have just tested a fresh installation of v0.8.0 on a raspberry 2 and opencpn works right. We should know what changes you have made from the original image. If you don't have changes, you are running a corrupted image. Try downloading again and passing the cheksum to the compressed file. Try to change the SD card too.

andyrbarrow, 2016-06-07:
Okay, I was thinking that was the next step! I'll give it a try. Thanks.

andyrbarrow, 2016-06-07:
Okay, mystery solved. What I thought was an RPI 2 was not (it was in a case and I couldn't read the card). Clearly OpenCPN didn't like trying to run with only 500MB of memory, shared with everything else. Once I extracted a REAL RPI 2 from another project and tried the card, everything worked fine, and much faster! 

Getting an RPI 3 in a couple of weeks, and I'm sure it's all going to run very well. Thanks for your help, and for putting together a great suite of software!

sailoog, 2016-06-08:
There is always a place for your old Pis :)

closed 2016-06-08 by sailoog