sailoog/openplotter #100: Minimal json strings via TCP or UDP?

joabakk, 2016-08-07:
Hi again,

Is is possible with the default settings to provide json deltas to the node server from openplotter? I have tried port 7777 and 5555 and 10110 (but that is nmea) but to no avail. Either it is not possible, or my delta has some fault in it:

{"context":"vessels.257817010","updates":[{"source":{"type":"NMEA0183","sentence":"RSA","label":"witosk","talker":"hum"},"timestamp":"","values":[{"path":"steering.rudderAngle","value":4.057890510886816}]}]}

sailoog, 2016-08-08:
Yes, in v0.8.0 there is a UDP provider defined for Op sensors in port 7777

I think you should provide a timestamp, try this:
`timestamp=str( datetime.datetime.utcnow().strftime('%Y-%m-%dT%H:%M:%S.%f') )[0:23]+'Z'`

sailoog, 2016-08-08:
Look at v0.9.0 branch is full of examples:

https://github.com/sailoog/openplotter/blob/v0.9.0/1w_d.py

joabakk, 2016-08-08:
Thanks a lot for the pointers

joabakk, 2016-08-08:
Thanks again, I got as far as to get the delta through to the server, and the debug message shows:
Could not parse JSON:'{"context":"vessels.257817010","updates":[{"source":{"type":"NMEA0183","sentence":"RSA","label":"witosk","talker":"hum"},"timestamp":"2016-08-08T15:06:18.163Z","values":[{"path":"steering.rudderAngle","value":4.057890510886816}]}]}'

In json representation, it shows:

`{
    "context": "vessels.257817010",
    "updates": [
        {
            "source": {
                "label": "witosk",
                "sentence": "RSA",
                "talker": "hum",
                "type": "NMEA0183"
            },
            "timestamp": "2016-08-08T15:06:18.163Z",
            "values": [
                {
                    "path": "steering.rudderAngle",
                    "value": 4.057890510886816
                }
            ]
        }
    ]
}
`

I just put a static timestamp in for now.

joabakk, 2016-08-08:
got it working now. timestamp and line terminator  did the trick!

closed 2016-10-13 by joabakk