sailoog/openplotter #104: V0.90 works great, but

joabakk, 2016-08-24:
Hi again,

I've had a chance to look at v0.9 and try it on board a vessel. In fact I had to leave it on board, they were so happy. A lot of cool and useful features, great work!

However I miss the nmea generation of at least the common nmea sentences like heading and true wind. Sk is the future, but I still use apps and instruments that don't know that. I agree that generating xdr sentences is wasted code, because they are so poorly defined and vary from system to system. And I like having everything in signal k so I can have the overview.

sailoog, 2016-08-25:
Be careful v0.9.0 is an alpha version and there are a lot of thinks that are not running yet. True wind and the rest of calculations is one of them. All data will be signal k but do not worry, we are implementing a NMEA generator in tools menu to be able to build even the most odd XDR sentence from signal k data :)

joabakk, 2016-08-31:
OK thanks for the warning.  I also like the diagnostics, and I am missing some conversions like Hz-RPM for engine RPM, J-Ah for battery banks, m3/s-l/h for consumption, ratio to percent, and rad/s-deg/min for ROT.

Where would I add that, to diagnostics only?

e-sailing, 2016-09-01:
diagnostics-SK-input.py start on line 199 (last commit) and unit-private.py on line 63.

Yes. It's only used by diagnostics at the moment.
- generating N2K sentences has a constant change.
- generating 0183 menu can + - * /  (does not work for Kelvin to Farenheit)
- apps working on SignalK should have there own converter
Be careful with RPM. If actisense reader is right, N2K (Canboat) to SignalK is converted to RPM not to HZ as it should. 
Generated NMEA0183 are send by udp 55565 (readme says 10110) at the moment.

sailoog, 2016-10-26:
closed via #119

closed 2016-10-26 by sailoog