sailoog/openplotter #109: Connecting to OpenPlotter via MQTT

toqduj, 2016-09-12:
Hi

I wanted to ask this in the forum, but I couldn't find that anywhere. 

I've built a simple GPS nodule by connecting an Adafruit Ultimate GPS to an Arduino MKR1000. That should connect over WiFi to a Raspberry Pi 3 running OpenPlotter 0.8.0. 

So far, I have the Arduino connecting to the Pi 3 with the wifi access point started from the OpenPlotter configuration panel. There, it is also indicated that there is an MQTT broker that I might connect to, with the IP addresses indicated on the right. There is no information on what (if any) access credentials would be required. 

However, trying to connect, nothing happens. I manually started a mosquitto broker via the terminal, and that works fine (so it looks like the port numbers for MQTT are not reserved by another process). 

So is it true that the MQTT broker needs to be started manually, that there is no configuration preset for that? Secondly, do I need to subscribe to that same MQTT server via the "MQTT" tab on the OpenPlotter configuration panel, to get any SignalK messages to OpenPlotter?.

Cheers, 

Brian.

sailoog, 2016-09-14:
Hi Brian,

OpenPlotter can work with 2 brokers, remote and local. If you provide 
name an pass in MQTT tab, you will start the local broker automatically. 
If you provide a remote address in MQTT tab, you will be able to sent 
and receive from that broker too. Both will share user and pass data and 
topics. The idea is to connect to the local broker when you are sailing 
and to connect to the remote broker at home. Sorry for the lack of 
documentation and thanks for testing OP!

Roberto.

El 12/09/16 a les 20:09, Brian R. Pauw ha escrit:

toqduj, 2016-09-14:
Got it, seems to connect now. What remains is for me to understand the rest of the signalk connection handshake protocol and implement it (http://signalk.org/developers/communication_protocols.html). So far, it seems like it's no problem to build the signalk json messages in the MKR1000: plenty of memory to go around. 

Thanks for the quick response, and the awesome work on OP!

toqduj, 2016-09-14:
P.S. when I start the local broker, do I need OP to subscribe to a signalk.put topic from the MQTT panel as well, or does that happen automatically?

closed 2016-09-25 by toqduj