sailoog/openplotter #112: Diagnostic with regular expressions

joabakk, 2016-09-20:
Hi again, on v0.9

Just connected most of the gear on board and set up with openplotter and a display

The signal k diagnostic is very useful, but unit and description is missing from electrical.batteries.x.voltage. I assume openplotter is not taking into account that there is a regexp involved?

Comments @tkurki?

tkurki, 2016-09-21:
My guess is that OpenPlotter uses literal lookup from the SK schema generated keyswithmetadata.json and does not handle the dynamic id (battery id) in the SK key.

Instead of having all SK clients have their own copy of schema data with units & enums & stuff it would probably be a good idea to implement metadata lookup so that the SK server provides the metadata.

See also 
- https://github.com/SignalK/specification/issues/244
- https://github.com/SignalK/specification/pull/240

sailoog, 2016-09-21:
Hi guys, let me check this tomorrow, no much time now, thanks!

closed 2017-08-27 by sailoog