sailoog/openplotter #113: Trouble sending MQTT SignalK message to OpenPlotter

toqduj, 2016-09-25:
Hi, 

I am trying to implement a WiFi connected GPS nodule, based on the Adafruit Ultimate GPS, and an Arduino/Genuino MKR1000 for the NMEA0183 -> signalK translation, and publishing to the OpenPlotter MQTT queue. The idea is that you can have independent deck- or mast-mounted GPS, compass and other nodules, while the OpenPlotter Raspberry Pi (3) is safely below deck. The nodules need only a power connection, but can contain a small battery as well. 

For now, I want to have a proof-of-concept implementation that connects to GPS and OpenPlotter, and sends a dummy SignalK message. This Arduino code is available here: https://bitbucket.org/toQDuj/signalk-gps-nodule

I tried my best with the MQTT implementation to stick to the following scheme: https://github.com/SignalK/specification/wiki/Communication-Protocols
However, lack of experience with MQTT and SignalK have now landed me at a roadblock: messages are not successfully sent to the SignalK server (started using this: https://github.com/sailoog/openplotter/issues/109). 

My debug messages are telling me only that the WiFi connection is set up, that I can connect to the MQTT broker, but not much more (I've modified the GPS coordinates, so the checksum is now incorrect...):

-->
Scanning available networks...
** Scan Networks **
number of available networks:14
[snip]
3) SignalK	Signal: -57 dBm	Encryption: WPA
[snip]
Attempting to connect to WPA SSID: SignalK
Waiting 1 seconds for network to settle
Attempting to connect to WPA SSID: SignalK
Waiting 1 seconds for network to settle
You're connected to the networkSSID: SignalK
BSSID: [snip]
signal strength (RSSI):-54
Encryption Type:2
IP Address: 10.10.10.204
MAC address: [snip]
Attempting MQTT connection...connected
Subscribing to ... signalk.put
{"self":"D344B1D0","version":"1","vessels":{"D344B1D0":{"uuid":"D344B1D0","environment":{"depth":{"belowTransducer":{"value":10.83,"timestamp":"2016-04-15T17:56:50.000Z","source":{"type":"NMEA0183","sentence":"DBT","label":"signalk/signalk-parser-nmea0183"}}}}}}}$PMTK001,220,3*30
$PMTK001,300,3*33
$PMTK001,314,3*36
$GPGGA,104712.000,5023.2238,N,01545.2436,E,1,07,0.98,84.9,M,44.8,M,,*58
$GPRMC,104712.000,A,5023.2238,N,01545.2436,E,0.21,205.96,250916,,,A*63
Successful GPS fix. Coordinates in D M.M format are:
  Latitude: 50 degrees, and 23.22 minutes North.
  Longitude: 15 degrees, and 45.24 minutes East.
$GPGGA,104715.000,5023.2238,N,01545.2436,E,1,07,0.98,84.9,M,44.8,M,,*59
$GPRMC,104715.000,A,5023.2238,N,01545.2436,E,0.14,149.20,250916,,,A*62
Sending SignalK Demo...
Failed to publish SignalK Demo
<--

If anyone can see what I'm missing, I'd be _very_ happy to hear about it. 

Cheers, 

Brian.

toqduj, 2016-10-03:
Tested on v0.9.0, same issue.

toqduj, 2016-10-09:
I've managed to modify some Freeboard classes to construct a fairly complete signalK message on the MKR1000 (of 3 kByte). This means there is nothing in the way of constructing wireless shipboard sensors in this manner. However, messages sent to the Openplotter MQTT broker from the MKR1000 still do not appear to be received or interpreted. Debug options on the broker, or a clearer schematic on how an MQTT broker is to be set up on Openplotter would be highly appreciated.

sailoog, 2016-10-10:
Hi Brian,
Sorry for the delay. MQTT messages are driven by Actions in openplotter. 

For 0.8.0:

- You have to set an Username and Password on tab MQTT. This will start the local broker, add an address if you are using a remote broker. 
- Add topics.
- Apply changes.
- Go to Actions tab.
- Add a trigger for the topic you have created, they will be at the end of the list.
- Select an operator and a value.
- Add an action for that trigger.

If you want to send MQTT messages for these topics there will be actions for them too.
For 0.9.0 it works almost the same but we are improving it.

toqduj, 2016-10-11:
(copy from the message on the SignalK list:)
Hi Sailoog,

Thanks for the response. So, just to make sure I'm not misunderstanding, this is a way to get messages from a wireless sensor into openplotter, right? All other examples I've seen so far using SignalK and MQTT are for sending status messages from openplotter to an external visualisation application. 

If this works, it would open up the gates for a whole bunch of wifi-connected sensors, so I'm quite keen on getting this off the ground.

toqduj, 2016-10-11:
Alright, message is coming through after adjustment of the buffer sizes on the Arduino MQTT software side. In the terminal window from which openplotter was started, I'm getting the following error now, everytime the status message is sent. This indicates that it's being received by SignalK, but there may be a problem with SignalK/Openplotter's JSON parser?

`
Could not parse JSON:{"context": "vessels.urn:mrn:imo:mmsi:00000000","updates":[{"source":{"type": "notifications","src":"mqtt"},"timestamp":"2016-10-11T19:29:25.429Z","values":[{"path":"notifications.mqtt.gpsput","value":{"message": "{"vessels":{"self":{"navigation":{"courseOverGroundTrue": {"value":0.0000000000},"courseOverGroundMagnetic": {"value":93.0000000000},"headingMagnetic": {"value":0.0000000000},"magneticVariation": {"value":0.0000000000},"headingTrue": {"value":0.0000000000},"pitch": {"value":0.0000000000},"rateOfTurn": {"value":0.0000000000},"roll": {"value":0.0000000000},"speedOverGround": {"value":0.0000000000},"speedThroughWater": {"value":0.0000000000},"state": {"value":"Not defined (example)"},"anchor":{"alarmRadius": {"value":0.0000000000},"maxRadius": {"value":0.0000000000},"position":{"latitude": {"value":0.0000000000},"longitude": {"value":0.0000000000},"altitude": {"value":0.0000000000}}},"position":{"latitude": {"value":0.0000000000},"longitude": {"value":0.0000000000},"altitude": {"value":0.0000000000}}},"alarm":{"anchorAlarmMethod": {"value":"sound"},"anchorAlarmState": {"value":"disabled"},"autopilotAlarmMethod": {"value":"sound"},"autopilotAlarmState": {"value":"disabled"},"engineAlarmMethod": {"value":"sound"},"engineAlarmState": {"value":"disabled"},"fireAlarmMethod": {"value":"sound"},"fireAlarmState": {"value":"disabled"},"gasAlarmMethod": {"value":"sound"},"gasAlarmState": {"value":"disabled"},"gpsAlarmMethod": {"value":"sound"},"gpsAlarmState": {"value":"disabled"},"maydayAlarmMethod": {"value":"sound"},"maydayAlarmState": {"value":"disabled"},"panpanAlarmMethod": {"value":"sound"},"panpanAlarmState": {"value":"disabled"},"powerAlarmMethod": {"value":"sound"},"powerAlarmState": {"value":"disabled"},"silentInterval": {"value":300},"windAlarmMethod": {"value":"sound"},"windAlarmState": {"value":"disabled"},"genericAlarmMethod": {"value":"sound"},"genericAlarmState": {"value":"disabled"},"radarAlarmMethod": {"value":"sound"},"radarAlarmState": {"value":"disabled"},"mobAlarmMethod": {"value":"sound"},"mobAlarmState": {"value":"disabled"}},"steering":{"rudderAngle": {"value":0.0000000000},"rudderAngleTarget": {"value":0.0000000000},"autopilot":{"state": {"value":"off"},"mode": {"value":"powersave"},"targetHeadingNorth": {"value":0.0000000000},"targetHeadingMagnetic": {"value":0.0000000000},"alarmHeadingXte": {"value":0.0000000000},"headingSource": {"value":"compass"},"deadZone": {"value":0.0000000000},"backlash": {"value":0.0000000000},"gain": {"value":0},"maxDriveAmps": {"value":0.0000000000},"maxDriveRate": {"value":0.0000000000},"portLock": {"value":0.0000000000},"starboardLock": {"value":0.0000000000}}},"environment":{"airPressureChangeRateAlarm": {"value":0.0000000000},"airPressure": {"value":1024.0000000000},"waterTemp": {"value":0.0000000000},"wind":{"speedAlarm": {"value":0.0000000000},"directionChangeAlarm": {"value":0.0000000000},"directionApparent": {"value":0.0000000000},"directionTrue": {"value":0.0000000000},"speedApparent": {"value":0.0000000000},"speedTrue": {"value":0.0000000000}}}}}}

Could not parse JSON:"}}]}]}
`

richardmbell, 2016-10-11:
I ran the output through a JSON checker/parser and there seems to be a problem with the JSON structure.
The output has an extra quote before the opening brace of “vessels”.

As output:
","values":[{"path":"notifications.mqtt.gpsput","value":{"message": "{"vessels":{"self":{"navigation":

Correct:
","values":[{"path":"notifications.mqtt.gpsput","value":{"message": {"vessels":{"self":{"navigation":

Looking at the source now.

Richard

toqduj, 2016-10-12:
Hi Richard. 

Actually, the quote marks are around the entire message that I sent over MQTT. The first section 
`Could not parse JSON:{"context": "vessels.urn:mrn:imo:mmsi:00000000","updates":[{"source":{"type": "notifications","src":"mqtt"},"timestamp":"2016-10-11T19:29:25.429Z","values":[{"path":"notifications.mqtt.gpsput","value":{"message": "`

and the last section 

`Could not parse JSON:"}}]}]}`

(minus the 'could not parse JSON:') have been added by openplotter. So either the quote marks are added by openplotter, or are part of the message I sent. I'll see what I can do to improve this.

sailoog, 2016-10-12:
The extra quotation is added by openplotter and you get the error because it is not ready to parse signal k/json yet.

I see that you are using OP v0.9.0 and it works a little bit different than v0.8.0. 

OP v0.9.0 uses signal k as main system, that means that all data will be stored in SK data model, MQTT messages included. We are doing a non orthodox use of notifications signal k group to do this and it works pretty well:

notifications.mqtt.topic.message=string/number

We have conceived MQTT in openplotter as a way of get string or numbers to be used as triggers for actions or values to update gauges or values to be send by SMS, email, MQTT... but you are trying to use MQTT messages to send signal k delta messages and update other groups different than notifications group.

OP is ready to get external signal k delta messages to feed the signal k database by UDP 55559 for serial devices and UDP 55561 for network devices but not by MQTT.

You could use these UDP inputs or you could wait for us to implement the MQTT signal k "parser", this is one of the planned improvements. The idea is to set the topic as "string/number" or "signal k delta" when you create it in openplotter. If it is defined as "signal k delta", it  would be sent to UDP 55561 to be parsed as normal signal k delta data and if it is defined as "string/number" it would be included in notifications.mqtt.topic.message=string/number to be used as trigger for actions or update gauges, etc.

toqduj, 2016-10-12:
Thanks for the quick reply.
If I understand it right, that means you are deviating a little from the SignalK proposed MQTT model regarding the signalk.put topic? see: https://github.com/SignalK/specification/wiki/Communication-Protocols

The way forward for me (UDP or wait for OP development) depends on the expected timescale for the implementation of the signal k delta parser in openplotter. If it's going to be there within about four months, it will be well in time for any trips I've got planned. MQTT does seem to be the neatest way of sending messages from sensor modules to the openplotter server (at least to my untrained eyes), but please correct me if simple UDP messages are a better way of doing things.

There still remains a lot of work to be done to make the GPS Wifi module code anywhere near useful, for which I think I'll need (at least) the winter months. For example, I'll likely 

- need to implement the correct connection model (http://signalk.org/specification/master/connection.html) using mdns discovery to discover the openplotter server. 

- Additionally, I need to more closely follow or fill in Rob's freeboard signalk data model. 

- Thirdly, I need to consider making some more wifi modules, perhaps a compass/IMU and a display module would be nice. 

Anyway, I think that makes this issue "closed", or "pending" for now, right?

sailoog, 2017-03-06:
I am working on this now. Reopening...

toqduj, 2017-03-10:
Awesome, looking forward to the result!

sailoog, 2017-03-15:
Solved by https://github.com/sailoog/openplotter/commit/23581089111e98e233a2f835859a3ac041c25e14

Now you can define topics as:

**General**: published data on this topic will be stored at "notifications.topic"

**Signal K key**:  published data on this topic will be stored at the selected signal k key. ex. "propulsion.engine1.Temperature"

**Signal K delta**:  published data on this topic will be parsed into signal k full model.

closed 2017-03-15 by sailoog

toqduj, 2017-04-29:
Changes to the handling worked wonderfully, MQTT SignalK updates are being received awesomely by the server!