sailoog/openplotter #118: Crash in mqtt_d after two days

toqduj, 2016-10-23:
After two days of sending (uninterpreted) MQTT messages to OpenPlotter, I get the following crash in mqtt_d.py. After restarting openplotter, message reception is resumed. I'm not sending unicode from the GPS sensor, so there must be a false byte generated somewhere. 

I'm not sure if incoming signalk messages should be allowed to contain unicode or not. For future language compatibility it would be very nice, but we'd overlook false byte generation such as this case, and run into issues when this value is passed further down the line.

I can't immediately tell if the non-ascii character is in the MQTT message (around the value), or in the complete SK message (somewhere around the timestamp), but it's probably a good idea to either/and: 1) sanitise the input (verify that the incoming message can be cast into ascii or unicode) or 2) encapsulate the message interpreter in a try/except block, so that faulty incoming messages don't cause OP to stop functioning. 

Anyway, the error is:
-->
Traceback (most recent call last):
  File "/home/pi/.config/openplotter/mqtt_d.py", line 97, in <module>
    client_local.loop_forever()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1261, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 811, in loop
    rc = self.loop_read(max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1073, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1475, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1943, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2118, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2274, in _handle_on_message
    self.on_message(self, self._userdata, message)
  File "/home/pi/.config/openplotter/mqtt_d.py", line 29, in on_message
    SignalK='{"context": "vessels.'+uuid+'","updates":[{"source":{"type": "notifications","src":"mqtt"},"timestamp":"'+timestamp+'","values":[{"path":"'+path+'","value":'+value+'}]}]}\n'
UnicodeDecodeError: 'ascii' codec can't decode byte 0xde in position 124: ordinal not in range(128)
<--

Cheers, 

B.

sailoog, 2016-10-24:
Ok, thanks for reporting.

toqduj, 2017-01-13:
Might've been a problem in the timestamp messages generated by the sensor, I noticed the odd character in the debug screen. Input validation on the MQTT incoming messages might be worthwhile, but I guess the issue can be closed for now.

sailoog, 2017-03-15:
closing, reopen if necessary.

closed 2017-03-15 by sailoog