sailoog/openplotter #120: Critical security issue with default password-less, encryption-less VNC server

toqduj, 2016-10-26:
I am not 100% sure, but I think Openplotter comes default with VNC enabled, but without a VNC password set, nor encrypted traffic. For the first, there is a checkbox in the OP settings GUI, but not for the second.

This default setting means that _anyone_ with an iPad in the Marina, and with absolutely no experience whatsoever, can access your boat's openplotter server. Since you're usually logged in as user "pi", this means you also have root access. In terms of security, this is the absolute worst. 

At least switch the VNC password protection on per default, and encrypt the connection for good measure as well. 

I also note from a port scan that there is a microsoft RDP server running. I haven't had the chance to check that for security issues, but please make sure that it's at least modestly secure!

sailoog, 2016-10-26:
Yes, this should be fixed. We should encourage users to change the pi user password when installing.
If OP is acting as access point it is not a problem because you have to know the network password but if it is connected to the marina router and you do not set a VNC password, anyone could find out your IP and connect by remote desktop. With RDP is even worst because if you do not change the pi user password anyone could connect by remote desktop too. I propose:

Encourage people to change the pi user from Raspberry Pi Configuration menu. Maybe we should check this and force them before starting OP.

Disable VNC by default and force user to set a password when enabling.

toqduj, 2016-10-27:
Sounds like a decent plan, not sure the "no encryption" should be a
user-accessible option at all, since the VNC clients I am familiar with all
can store passwords.

Setting passwords sounds like something that could be part of a OpenPlotter
set-up script or dialog. Best is if you can add a memorable
password-generator to provide a random decent password as a starting value.

Apart from the gaping security hole, I love the work so far! :).

Cheers,

B.

sailoog, 2016-12-15:
Closed by https://github.com/sailoog/openplotter/commit/16d57912bfc51b4104ca0e92c210b1eed874b974

toqduj, 2016-12-30:
Still an issue in the latest v0.9.0. Also, perhaps this can be side-stepped. I noticed that Raspbian Jessie now comes with a licensed VNC client already installed, which, if I'm not misled by another installation, uses an encrypted connection. This means that Openplotter doesn't need to install this anymore, just switch it on.

sailoog, 2016-12-30:
All password issues have been solved in 0.9.0 but you are right that we forgot encrypted connection. We really need/want it?

Raspbian includes now a VNC client and a VNC server. We keep the client but uninstall the server to make it compatible with xrdp, look at this: https://www.raspberrypi.org/blog/introducing-pixel/
> Please note that if you already use xrdp to remotely access your Pi, this conflicts with the RealVNC server, so you shouldn’t install both at once.

So we keep the x11vnc server which is compatible with xrdp.

Reopening issue to study the case.

Thanks

sailoog, 2016-12-30:
Do you know if encryption affects performance?

toqduj, 2016-12-30:
I may have been mistaken in assuming the no-password login still existed,
my VNC client connected without prompting, but (come to think of it) it may
have remembered the previous password. I will verify in the next few days,
sorry about that.

I have not noticed any lag when accessing or operating a Raspberry Pi using
the RealVNC server with added encryption. The client was a 5 year old iPad
2. I don't think it is a major resource hog. I have not compared the
increased load on the server, however.

Too bad the included RealVNC cannot be used, it would have saved some
effort (and XRDP is much less responsive on the iPad, so I don't use it
anyway). fewer tools to deploy, fewer things to go wrong.

Apropos, I'm very much looking forward to the upcoming reimplementation of
the MQTT server :).

Cheers,

B.

sailoog, 2016-12-30:
Now there is a checking at startup and if you have default passwords for pi user and wifi access point or you have not set a vnc password, you will get a warning.

I will make some test with encryption and I will set it by default.

Since there is not agreement on signal k specification forum about MQTT I am working on an all in one solution to manage delta inputs, feed signal k keys directly and allow general use of topics ;) I will publish on the open issue for this when is ready to test.

toqduj, 2016-12-30:
Hmm, I might have gotten the warning, but VNC is then still automatically
started despite the security issues, no? That means that the unfamiliar
user will - per default - have either VNC running with the standard pi
password, or no password at all.

As for the MQTT implementation, it sounds like you're building a signal K
multiplexer! Looking forward to having it.

Cheers,

Brian.

toqduj, 2017-01-01:
Looks like I might have been mistaken regarding the password. Encrypted connection would still be nice, though!

B.

toqduj, 2017-01-07:
Hi @sailoog, did you notice any performance issues with encryption on?

B.

toqduj, 2017-01-18:
I checked again on a fresh 0.9.0 install: VNC is enabled with the password disabled. A warning message about the password pops up on the startup script.

sailoog, 2017-05-08:
v0.10.0 is managing raspbian, wifi and vnc passwords at startup, should we close this issue?

toqduj, 2017-05-09:
Still a little unsettling that the VNC password is blank on startup, but ok...

closed 2017-05-09 by toqduj