sailoog/openplotter #122: NMEA 0183 additional UDP input not working ?

sushicodeur, 2016-11-11:
Hello,

Well I'm trying to send udp messages to my raspberry pi with OpenPlotter 0.8.0 noobs.

So I declared an additional Input in OpenPlotter like so : 
arduino | UDP | 127.0.0.1 | 10112 | none | nothing

But after applying changes and restart kplex (and even restarting the raspberry, it looks like nothing is listening to this port. netstat -lptn shows kplex listening to 10110 but nothing on 10112.

I also edited .kplex.conf and set optional=no but still no luck.

Am I missing something ? Is there some kind of restriction on the number of interfaces on a same ip address ?

Many thanks

e-sailing, 2016-11-13:
Check if you get anything "nc -ul 127.0.0.1 10112". Why don't you send it to 10110?

sushicodeur, 2016-11-14:
Hello, 

Thanks for the hint. Well I tried to create a new connection as I didn't manage to use the 10110, I thought it might be used by some processes... So I'll try to use the 10110 instead.

So far, if i try `nc -ul 127.0.0.1 10110` and on my remote computer I send `echo -n "\$WIXDR,P,1.0194,B,BARO*4F" > /dev/udp/192.168.1.124/10110` it doesn't see anything coming. (192.168.1.124 is the local ip of the raspberry).

I have to do `nc -ul 0.0.0.0 10110` to see the packet.

But in both case the inspector don't seems to get the message.

Have you got example of how to send NMEA sentence to the raspberry from network ?

Thanks

sushicodeur, 2016-11-22:
Hello,

So I've made some progress, and yes, it looks like any additional UDP input defined in OpenPlotter's interface don't work for me.

What I have done :

* An arduino sends over wifi nmea sentences (`$IIMTA,25.6,C*04` and `$WIMDA,29.984,I,1.015,B,25.6,C,15.0,C,,,,,172.0,T,173.0,M,6.2,N,3.2,M*7D`), using UDP to the IP 192.168.0.255:10112 (so this is broadcasted to all my local network)
* On my desktop computer, with openCpn, having an UDP connection on 10112, I get the air temperature of 25.6 and the baro at 1015 in the dashboard
* Now on the Pi, when I add a network connection in OpenPlotter with the same parameters, OpenCPN doesn't get the info, nor the inspector
* If I remove this connection in OpenPlotter and add one in OpenCPN then I get my data, but I'd like to benefit of kplex...

The following command does the same as my arduino does : `sudo  echo -n "\$IIMTA,25.6,C*04" | socat - UDP-DATAGRAM:192.168.0.255:10112,broadcast` does it works for you ?

If you need any other information to test this, just ask :)

Thank you very much

e-sailing, 2016-11-25:
I have the same problem doing it your way. I can't tell you why it doesn't work. In python we use udp with port 10110 no problem. But if I send it by your sentence with 10110 it doesn't work. As you already mentioned it has to do with kplex.
In v0.9.0 you can send your input to signalk (in signalk format) and convert it to nmea 0183 or 2000. Have a look at https://github.com/e-sailing/esp8266_2_signalk. Don't invest to much time into this and wait for the v0.9.0 image or search for a solution on http://www.stripydog.com/kplex.

sushicodeur, 2016-11-25:
Ok, thank you very much for putting time on this. I can wait for 0.9.0 to be out. If I get kplex to work, I'll let you know here.

Thanks again

closed 2016-11-25 by sushicodeur