sailoog/openplotter #123: How do i publish SK data to a mqtt topic 0.90

dmdelorme, 2016-11-14:
HI i had mqtt working in 0.80 but i am at a loss on how to publish via actions a SK value eg. environment.water.temperature.value. 0.90 looks awesome. I  wish the nmea 0183 generator allowed for a few more significant digits. in the format ie x.xxxxx

dmdelorme, 2016-11-21:
the latest clone of opencpn solved some issues with pressure and nmea sensor messages. no more MDA messages need
I also change the following in SK-base_d i should change the code associated to x.xxxx or create a new format float X.XXX
if i[1] == 'x.x':
					value_str = str(round(value, 1))
				elif i[1] == 'x.xx':
					value_str = str(round(value, 6))
I am still at a loss on how to Access SK values in actions for MQTT

sailoog, 2016-11-21:
We are working on MQTT actions but there is a long discussion on this concept in the signalk specification and we are waiting for conclusions to implement it on 0.9.0
https://github.com/SignalK/specification/issues/291

closed 2017-08-27 by sailoog