sailoog/openplotter #125: Port 5555

dmdelorme, 2016-11-20:
When i open http://openplotter:5555. It looks like SignalK 
{"context":"vessels.urn:mrn:imo:mmsi:1999999","updates":[{"source":{"type":"I2C","src":"LSM9DS1","label":"OP_sensors"},"timestamp":"2016-11-20T20:34:57.500Z","values":[{"path":"navigation.headingMagnetic","value":3.52629010044},{"path":"navigation.attitude.roll","value":-0.00658904360104},{"path":"navigation.attitude.pitch","value":-0.0271246105443}]}]}...... and more
Where is this in the code?

sailoog, 2016-11-21:
https://github.com/sailoog/openplotter/blob/v0.9.0/i2c_d.py#L205

dmdelorme, 2016-11-22:
i did a search after not finding it in the Openplotter code.
grep -rnw '.config/' -e "5555"
the results were.
.config/signalk-server-node/lib/interfaces/tcp.js:24:  var port = 5555;
.config/signalk-server-node/lib/interfaces/tcp.js:66:    port: 5555

sailoog, 2016-11-22:
Ops, you are right. It seems that the last version of signalk-server-node creates a TCP output in 5555 by default. there you will have a stream of delta messages with data updates. thanks for notifying.

closed 2016-12-03 by dmdelorme