sailoog/openplotter #126: Charts are not loading.

addboy58, 2016-11-25:
I have used the chart down loader and scannded the charts, but not charts load on the map. What do I need to do to fix this?
thx
D

Rodin1, 2017-01-16:
I used this link and run perfectly:

http://wiki.openstreetmap.org/wiki/DE:KAP-charts_from_OpenSeaMap#Download

;)

roland57, 2017-01-29:
Hi friends, 
the download is quite easy but the compressed format is not readable for OpenCPN. You have to decompress them. There is no tool delivered with OpenPlotter to do that. i case I just didn't find it Would appreciate a comment.

I have installed p7zip from the command line: 

sudo apt-get update
sudo apt-get install p7zip

When p7zip is installed go to the directory where the downloaded file is stored, then type: 

p7zip -d    Filename       (where 'filename' is the name of the downloaded file.)

I was not able to install p7zip using the Application manager in the GUI in OpenPlotter.

Hope this helps.
Roland

toqduj, 2017-05-09:
I've also had no problem with the OpenSeaMap charts. Problem might be one of documentation: charts need to be decompressed, placed in a suitable directory, and OpenCPN needs to be told to rebuild the chart database. More of an OpenCPN documentation issue than an OpenPlotter one.

closed 2017-08-27 by sailoog