sailoog/openplotter #127: openplotter 0.90 failts to open

dmdelorme, 2016-11-29:
I am getting this back for openplotter
pi@openplotter:~ $ openplotter 
Gtk-Message: Failed to load module "canberra-gtk-module"
Traceback (most recent call last):
  File "/home/pi/.config/openplotter/openplotter", line 3137, in <module>
    MainFrame().Show()
  File "/home/pi/.config/openplotter/openplotter", line 77, in __init__
    self.vessel_self = checkVesselSelf()
  File "/home/pi/.config/openplotter/classes/check_vessel_self.py", line 71, in __init__
    self.mmsi = data['mmsi']
KeyError: 'mmsi'
I am at a loss as to what is causing this
this is new install based on your directions

sailoog, 2016-11-30:
I think we have fixed this, try to download again 0.9.0 branch

dmdelorme, 2016-12-03:
Awesome it is all good now

closed 2016-12-03 by dmdelorme