sailoog/openplotter #132: MPU-9255

dmdelorme, 2016-12-03:
I am trying to get the MPU-9255 working i have recompiled RTIMULib Nick-Currawong GIT. I can see the device and read data from it but it seems very unstable and has an issue with lag. Has anyone got this working or should i just find a better or more supported Chip ie 9250. Open to suggestions.

seandepagnier, 2016-12-24:
are you sure about lag?  I have it too, try my fork which incorporates other stuff

https://github.com/seandepagnier/RTIMULib2

dmdelorme, 2017-01-03:
I had down loaded your RTIMU before but i will try this again and set up some debugging stuff to find were the lag is from I think it is something in Openplotter. I have 3 IMU now a 9250, 9255 and LSM9DS1. All of them have the same issue.

closed 2017-08-09 by sailoog