sailoog/openplotter #137: Error on parse when creating port with other language than English

RataDP, 2016-12-26:
When I create a new port when I have the openplotter in Spanish, the port does not recieve anything. I checked the configuration file and it has the following options. The option for the port was filter anything.
```
[udp]
name=portspanish
direction=in
optional=yes
address=127.0.0.1
ifilter=nada <--
ofilter=nada <--
```
It happends on TCP, UDP and Serial. In the English version when I created another port with the same options, _filter anything_, the configuration file was:
 
```
[udp]
name=portenglish
direction=in
optional=yes
address=127.0.0.1
```

So when the openplotter starts on console, it writes `Error on parse config file in line 36`, line 36 is `ifilter=nada`. When I removed this port openplotter works fine.

I will work on a PR to solve this problem. But for now, the *workaround* is create the port in English version and then switch to Spanish o Catalonian.

**Tested with Spanish and Catalonian**
The openplotter is from the **pi image version 9.0 with NOOBs**

sailoog, 2016-12-26:
Oops, big bug. I don't understand how it hasn't been detected before. Out of the work now, I will try to solve this tomorrow. I'm afraid lots of multilingual issues will appear in 0.9.0 yet.

Moltes gracies!!

RataDP, 2016-12-26:
@sailoog De res ^^.

Your welcome. Just now I'm reading your code to see where this issue is happening.
If you need something to test, just tell me.

nicolas-j, 2016-12-27:
Hi,

 I'm usualy a bit shy about non working things on my setup, but just in
case it would occur to others, I could not get my gps back to work with my
latest update (noobs Op, update from github, adafruit gps on ttys0, 9600
Bauds).
Work fine with the same hardware, regular Jessie,  Op 0.9 separate
installation.
Node-red + ti sensortag cc2650 = fantastic résults !

Nicolas.

closed 2017-05-08 by sailoog