sailoog/openplotter #145: Cannot share huawei 4g mifi mobile to WiFi hotspot

snowboardergrun, 2017-02-07:
When i connect the huawei E5373 to openplotter i cannot choose it in "sharing internet devices"
my  3g huawei 169g internet dongel is working fine.

when i do ifconfig i get

3g huawei 169g   i get "ppp0"  is oke i can choose it
4g huawei 5373   i get "wwan0" is not visible

is it posible to add this "wwan0"device ?

sailoog, 2017-02-07:
You can try to set it manually in code. Let me know what OP version are you running and I will guide you.

snowboardergrun, 2017-02-07:
i run the latest 9.0 version

sailoog, 2017-02-08:
Open the file /home/pi/.config/openplotter/openplotter
and after the line 671: https://github.com/sailoog/openplotter/blob/v0.9.0/openplotter#L671
insert this line :

			if 'wwan' + ii in output: self.available_share.append('wwan' + ii)

respect the tab spaces at the beginning. You should see this:

			if 'eth' + ii in output: self.available_share.append('eth' + ii)
			if 'ppp' + ii in output: self.available_share.append('ppp' + ii)
			if 'usb' + ii in output: self.available_share.append('usb' + ii)
			if 'wwan' + ii in output: self.available_share.append('wwan' + ii)

Let me know if this works and I will merge it into development branch.
thanks

snowboardergrun, 2017-02-09:
Thanks , It's okay now

sailoog, 2017-02-10:
fixed by https://github.com/sailoog/openplotter/commit/85c518e2a090311518714c7bc1a7447b85d08a61

closed 2017-02-10 by sailoog