sailoog/openplotter #149: offset does not change values

dmdelorme, 2017-04-05:
I am a loss with this one i have checked openplotter.conf it is the same as the Git. open plotter writes the value to the config and reads the value but it does not change the value for the sensors.. I am on the current beta.

sailoog, 2017-04-05:
Where? 1W, I2C, SPI?

dmdelorme, 2017-04-05:
Woops i2c

dmdelorme, 2017-04-05:
**Totally off target.**
But I just realized that if i lived in the EU it would be Payday $$$$
 Sorry I am a Canadian

dmdelorme, 2017-04-06:
When I try to Update to today’s changes i get the following on start up

> startup restart
Gtk-Message: Failed to load module "canberra-gtk-module"
Signal K starting
No settings/defaults.json available
signalk-server running at 0.0.0.0:3000

cat: /dev/ttyOP_SKVA: No such file or directory

Cannot open NGT-1-A device /dev/ttyOP_N2K

GET /signalk/v1/api/vessels/self 200 35.595 ms - 19
/home/pi/.config/signalk-server-node/lib/pipedproviders.js:47
          update.source.label = providerConfig.id;
                              ^

TypeError: Cannot set property 'label' of undefined
    at /home/pi/.config/signalk-server-node/lib/pipedproviders.js:47:31
    at Array.forEach (native)
    at FromJson.<anonymous> (/home/pi/.config/signalk-server-node/lib/pipedproviders.js:46:22)
    at emitOne (events.js:96:13)
    at FromJson.emit (events.js:188:7)
    at readableAddChunk (_stream_readable.js:176:18)
    at FromJson.Readable.push (_stream_readable.js:134:10)
    at FromJson.Transform.push (_stream_transform.js:128:32)
    at FromJson._transform (/home/pi/.config/signalk-server-node/providers/from_json.js:35:10)
    at FromJson.Transform._read (_stream_transform.js:167:10)
[Errno 111] Connection refused

Stop Node-RED
 
Use   node-red-start   to start Node-RED again
 

Start Node-RED
 
Once Node-RED has started, point a browser at http://192.168.1.219:1880
On Pi Node-RED works better with the Firefox browser
 
Use   node-red-stop                          to stop Node-RED
Use   node-red-start                         to start Node-RED again
Use   node-red-log                           to view the recent log output
Use   sudo systemctl enable nodered.service  to autostart Node-RED at every boot
Use   sudo systemctl disable nodered.service to disable autostart on boot
 
To find more nodes and example flows - go to http://flows.nodered.org
No supported devices found.
 
pi@openplotter:~ $ Started Node-RED graphical event wiring tool..
6 Apr 18:26:52 - [info]
Welcome to Node-RED
===================
6 Apr 18:26:52 - [info] Node-RED version: v0.16.2
6 Apr 18:26:52 - [info] Node.js  version: v6.10.1
6 Apr 18:26:52 - [info] Linux 4.9.17-v7+ arm LE
6 Apr 18:26:54 - [info] Loading palette nodes
6 Apr 18:26:56 - [info] Dashboard version 2.3.6 started at /ui
6 Apr 18:26:58 - [info] Settings file  : /home/pi/.node-red/settings.js
6 Apr 18:26:58 - [info] User directory : /home/pi/.node-red
6 Apr 18:26:58 - [info] Flows file     : /home/pi/.node-red/flows_openplotter.json
6 Apr 18:26:58 - [info] Server now running at http://127.0.0.1:1880/
6 Apr 18:26:59 - [info] Starting flows
6 Apr 18:26:59 - [info] [udp out:sendSK] udp ready: localhost:55559
6 Apr 18:26:59 - [info] Started flows

dmdelorme, 2017-04-08:
I updated SignalK fixed the above. But i am not able to get data from I2C devices and I am getting the following.. 
ERROR:websocket:error from callback <function <lambda> at 0x7556c630>: 'source'
ERROR:websocket:error from callback <function <lambda> at 0x7556c630>: 'source'
ERROR:websocket:error from callback <function <lambda> at 0x7556c630>: 'source'
ERROR:websocket:error from callback <function <lambda> at 0x7556c630>: 'source'
ERROR:websocket:error from callback <function <lambda> at 0x7556c630>: 'source'
ERROR:websocket:error from callback <function <lambda> at 0x7556c630>: 'source'
.....

dmdelorme, 2017-04-11:
my node-red flow is finding all the sources and values but openplotter diagnosis SK input not showing them ie any i2c, 1W... but it is in SignalK
debug output from Node-red

`2017-04-11, 5:11:18 AMnode: d32acde9.67e0cmsg.payload : string[319]
"{"updates":[{"$source":"I2C.LSM9DS1","values":[{"path":"navigation.headingMagnetic","value":1.67222254107},{"path":"navigation.attitude.roll","value":-0.00149698586627},{"path":"navigation.attitude.pitch","value":-0.0375929316181}],"timestamp":"2017-04-11T10:11:18.824Z"}],"context":"vessels.urn:mrn:imo:mmsi:98765431"}"
2017-04-11, 5:11:18 AMnode: d32acde9.67e0cmsg.payload : string[190]
"{"updates":[{"$source":"1W.1wtemp","values":[{"path":"environment.water.temperature","value":271.025}],"timestamp":"2017-04-11T10:11:18.834Z"}],"context":"vessels.urn:mrn:imo:mmsi:98765431"}"
**2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[236]
"{"updates":[{"source":{"type":"cabinTEMP","src":"OP2","label":"wifi"},"values":[{"path":"environment.inside.mainCabin.temperature","value":293.025}],"timestamp":"2017-04-11T10:11:19.188Z"}],"context":"vessels.urn:mrn:imo:mmsi:98765431"}"**
2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[379]
"{"context":"vessels.urn:mrn:imo:mmsi:98765431","updates":[{"source":{"label":"kplexOutput"},"timestamp":"2017-04-11T10:11:19.326Z","values":[{"path":"navigation.position","value":{"source":{"type":"NMEA0183","sentence":"RMC","label":"signalk-parser-nmea0183","talker":"GP"},"timestamp":"2017-05-11T10:11:18.000Z","longitude":-97.12520833333333,"latitude":49.905748333333335}}]}]}"
2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[361]
"{"context":"vessels.urn:mrn:imo:mmsi:98765431","updates":[{"source":{"type":"NMEA0183","sentence":"RMC","label":"kplexOutput","talker":"GP"},"timestamp":"2017-05-11T10:11:18.000Z","values":[{"path":"navigation.courseOverGroundTrue","value":1.6704546270837726},{"path":"navigation.speedOverGround","value":0},{"path":"navigation.magneticVariation","value":0}]}]}"
2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[319]
"{"updates":[{"$source":"I2C.LSM9DS1","values":[{"path":"navigation.headingMagnetic","value":1.67187802622},{"path":"navigation.attitude.roll","value":-0.00181724163146},{"path":"navigation.attitude.pitch","value":-0.0382660170877}],"timestamp":"2017-04-11T10:11:19.337Z"}],"context":"vessels.urn:mrn:imo:mmsi:98765431"}"
2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[368]
"{"context":"vessels.urn:mrn:imo:mmsi:98765431","updates":[{"source":{"type":"NMEA0183","sentence":"VTG","label":"kplexOutput","talker":"GP"},"timestamp":"2017-04-11T10:11:19.000Z","values":[{"path":"navigation.courseOverGroundMagnetic","value":0},{"path":"navigation.courseOverGroundTrue","value":1.6704546270837726},{"path":"navigation.speedOverGround","value":0}]}]}"
2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[379]
"{"context":"vessels.urn:mrn:imo:mmsi:98765431","updates":[{"source":{"label":"kplexOutput"},"timestamp":"2017-04-11T10:11:19.577Z","values":[{"path":"navigation.position","value":{"source":{"type":"NMEA0183","sentence":"GGA","label":"signalk-parser-nmea0183","talker":"GP"},"timestamp":"2017-04-11T10:11:00.000Z","longitude":-97.12520833333333,"latitude":49.905748333333335}}]}]}"
2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[457]
"{"context":"vessels.urn:mrn:imo:mmsi:98765431","updates":[{"source":{"label":"kplexOutput"},"timestamp":"2017-04-11T10:11:19.579Z","values":[{"path":"navigation.gnss","value":{"source":{"type":"NMEA0183","sentence":"GGA","label":"signalk-parser-nmea0183","talker":"GP"},"timestamp":"2017-04-11T10:11:00.000Z","quality":2,"satellites":10,"antennaAltitude":229,"horizontalDilution":0,"geoidalSeparation":-24,"differentialAge":0,"differentialReference":0}}]}]}"
2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[236]
"{"updates":[{"source":{"type":"cabinTEMP","src":"OP2","label":"wifi"},"values":[{"path":"environment.inside.mainCabin.temperature","value":293.025}],"timestamp":"2017-04-11T10:11:19.601Z"}],"context":"vessels.urn:mrn:imo:mmsi:98765431"}"
2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[379]
"{"context":"vessels.urn:mrn:imo:mmsi:98765431","updates":[{"source":{"label":"kplexOutput"},"timestamp":"2017-04-11T10:11:19.727Z","values":[{"path":"navigation.position","value":{"source":{"type":"NMEA0183","sentence":"RMC","label":"signalk-parser-nmea0183","talker":"GP"},"timestamp":"2017-05-11T10:11:19.000Z","longitude":-97.12520833333333,"latitude":49.905748333333335}}]}]}"
2017-04-11, 5:11:19 AMnode: d32acde9.67e0cmsg.payload : string[361]
"{"context":"vessels.urn:mrn:imo:mmsi:98765431","updates":[{"source":{"type":"NMEA0183","sentence":"RMC","label":"kplexOutput","talker":"GP"},"timestamp":"2017-05-11T10:11:19.000Z","values":[{"path":"navigation.courseOverGroundTrue","value":1.6704546270837726},{"path":"navigation.speedOverGround","value":0},{"path":"navigation.magneticVariation","value":0}]}]}"
2017-04-11, 5:11:20 AMnode: d32acde9.67e0cmsg.payload : string[368]
"{"context":"vessels.urn:mrn:imo:mmsi:98765431","updates":[{"source":{"type":"NMEA0183","sentence":"VTG","label":"kplexOutput","talker":"GP"},"timestamp":"2017-04-11T10:11:19.000Z","values":[{"path":"navigation.courseOverGroundMagnetic","value":0},{"path":"navigation.courseOverGroundTrue","value":1.6704546270837726},{"path":"navigation.speedOverGround","value":0}]}]}"
2017-04-11, 5:11:20 AMnode: d32acde9.67e0cmsg.payload : string[318]
"{"updates":[{"$source":"I2C.LSM9DS1","values":[{"path":"navigation.headingMagnetic","value":1.67112557716},{"path":"navigation.attitude.roll","value":-0.00208081710319},{"path":"navigation.attitude.pitch","value":-0.037662754736}],"timestamp":"2017-04-11T10:11:19.826Z"}],"context":"vessels.urn:mrn:imo:mmsi:98765431"}"`
it appears the the context is missing i have another pi with an older version outputting Maincabin temp WIFI.OP2 from a 1w temp sensor.

sailoog, 2017-04-11:
@dmdelorme as you have noticed by this issue: https://github.com/SignalK/signalk-server-node/issues/101, lots of parts of openplotter are not working due to latest changes on signal k. 

I have located most of the problems and working on it now. Download the latest image 0.10.0 and let's work from there. Thanks.

http://www.sailoog.com/en/blog/download-openplotter-rpi-v0100alpha-noobs

dmdelorme, 2017-04-11:
Ok will try I have a few bits and pieces  to deal with but I will redo the boobs but I am running kernel 4.9 as some other hardware needs it.. the project is looking awesome.. the new kernel has corrected an issue with sensor noise I am getting less then .1deg noise on my IMU now

dmdelorme, 2017-04-11:
I ment noobs autocorrect

dmdelorme, 2017-04-11:
version 0.10.1 Awsome work most of the issues fixed still not changing values for i2c but reponse time and lagging seems fixed for Pitch and roll in Node-red. SignalK diag is working... I will poke around a bit tomorrow I am so happy i could go out and buy a boat but the lake still has several feet of ice still.. 
What do you use for a dev environment ie debugger? I would like to contribute. ie calculations for heave, etc. I will see if i can find more info with the issue with i2c not loading adjustments. but this is not bad as it is outputting true level it will need to be fixed as i will have to level the device once mounted on a boat.
I live near Lake Winnipeg and i would like to set my boat up to do mapping.. ie Depth... etc as my government only sells maps in BSBv4 and they are very poor and incomplete.

closed 2017-04-12 by dmdelorme