sailoog/openplotter #153: IMU - I2c - 0.11.3 alpha

ygageot, 2017-04-29:
I just installed an IMU with MPU-9250 and MS5611 with 0.11.3 alpha.

With the RTIMULibDemoGL I see the real time values and OpenGL and I made the calibration.
I have the status : _access in use, mag in use, runtime mag not in use, ellipsoid not in use_
In the SK diagnostics I do not see any value from I2C like navigation.attitude.roll,...

Is there something to activate ?

ygageot, 2017-04-29:
news : I have now in SK pressure and temperature from OPsensors.I2C.MS5611 source
**and nothing like heading roll and pitch from MPU-9250**

sailoog, 2017-05-02:
I2C sensors are still buggy, working on it now. They will be fixed on next update. New sensors supported and better interface to make possible adding future sensors.

infernix, 2017-05-05:
I've got an MPU9250 hooked up but:

```
          def work_imu():
                        print 'inside work_imu'
                        threading.Timer(rate_imu, work_imu).start()
                        if imu.IMURead(): <--- only fires once
```

So the data is stale; it only gets updated on the first run of `read_sensors_d.py`

Trying the RTIMULib[ Fusion.py](https://raw.githubusercontent.com/RPi-Distro/RTIMULib/b949681af69b45f0f7f4bb53b6770037b5b02178/Linux/python/tests/Fusion.py) works fine. I suspect a threading issue because I don't see any obvious bug in the imu code.

sailoog, 2017-05-06:
Yes, it was a threading issue, fixed on next update on Monday.

ygageot, 2017-05-06:
@sailoog Thanks

sailoog, 2017-05-08:
v0.11.4 is in the wild. I2C is working again. Added BME280 support (temp, hum, press). I think I have not tested calibration enough, please report any issue. Thanks.

ygageot, 2017-05-11:
With 0.11.4 alpha version it works for me.

For calibration and after saving min/max and having more than 200 counts per octant;
If I press the button **Process ellipsoid** I have the message in popup : **Failed to executeRTEllipsoidFit.m.  Only min/max calibration available.**

sailoog, 2017-05-12:
I will check, thanks. Anyone else?

infernix, 2017-05-12:
Didn't work for me yet after an upgrade; will have to test again when I have time. could be a while.

ygageot, 2017-05-13:
I checked with text calibration and I have this message on console 
no such file **/home.pi/.config/openplotter/imu/RTEllipsoidFit.m**

in fact the file is in **/home.pi/.config/openplotter/imu/RTEllipsoidFit/RTEllipsoidFit.m**

sailoog, 2017-05-15:
Yes, probably a path issue, thanks.

sailoog, 2017-06-01:
Ellipsoid calibration fixed on 0.11.9

Test please

closed 2017-08-09 by sailoog