sailoog/openplotter #154: MQTT deltas from WiFi GPS: arriving but not being added to SignalK model?

toqduj, 2017-04-29:
With the latest changes to OpenPlotter, the SignalK delta messages that are being sent by my Wifi GPS unit over MQTT, are being received by OpenPlotter. I'm very happy with the success so far. 

I suspect I am perhaps missing a step (or two), but I'm not seeing the received navigation data being interpreted, not by any of the signalk gauges, nor is it being forwarded to OpenCPN. Do I have to configure actions to get that information forwarded? Are the deltas perhaps not completely up to spec?

The SK deltas being sent are of the following format:
{
	"updates": [{
		"source": {
			"timestamp": "2017-04-29T16:41:48Z",
			"$source": "deck.gps001"
		},
		"values": [{
			"path": "navigation.speedOverGround",
			"value": 0.7665223
		}]
	}]
}

The OpenPlotter SignalK diagnostic shows the following (there are valid lat and lon)..
[image]
The SK deltas are not preceded by a "Hello" message. Not sure if required. 

Any insight by anyone on the issue would be very welcome!

B.

toqduj, 2017-05-01:
Small update: with the "instrument panel" browser app in the signalk node server, the values are shown. With any of the other browser apps, nothing is shown. Curious. 

I edited the sent JSON a bit to be closer to the OP document FAQ. I also try to resemble the SK delta as shown in the SignalK documentation, but when I put the timestamp outside the "source" information, it is not accepted by OP. So I keep it in there:

{
	"updates": [{
		"source": {
			"timestamp": "2017-05-01T14:52:13Z",
			"src": "deck.gps001",
			"type": "signalk-delta"
		},
		"values": [{
			"path": "navigation.courseOverGroundTrue",
			"value": 1.1114256
		}]
	}]
}

[image]

toqduj, 2017-05-01:
The node-red dashboard and freeboard show *some* of the information too, although there appears to be a radian / degree mixup in their COG pointers. Anyway, progress :).

sailoog, 2017-05-02:
Your delta message is not valid. It should be:
```
{
  "updates": [
    {
      "$source": "deck.gps001",
      "timestamp": "2017-05-01T14:52:13Z",
      "values": [
        {
          "path": "navigation.courseOverGroundTrue",
          "value": 1.1114256
        }
      ]
    }
  ]
}
```
There is a big difference between "source" and "$source". Source is for NMEA 1083 and 2000 data and contains sub keys like src, type, talker, sentence, pgn... but $source is for custom data and sensors. In your case you are passing NMEA data but it has already been parsed so you need to pass it as custom data. Using $source you can create a hierarchic structure like "deck.gps001", "deck.gps002"...

toqduj, 2017-05-03:
Thanks! I've updated the delta message to be identical to yours. My previous ones have been constructed based on the examples in the signalk documentation, but this example would be good to have there too... OpenCPN doesn't receive any positions yet, but that's a later issue. 

There is one issue with the deltas: when I put in the timestamp, which is constantly updated from the GPS time signal, the "diagnostic SignalK input" window only updates once (when opened), but does not update when the new messages come in (also indicates an interval of 0.00). When I remove the timestamp from the delta message, the diagnostic window continually updates. There might be a slight (few seconds) offset between raspberry Pi time and the GPS time, causing some discrepancy there. Or maybe it's because I'm sending five updates for five different parameters, all using the same timestamp. So I've removed the GPS timestamp for now. 

As indicated in the screenshot, the instrument panel shows two sets of everything. The no_source ones are updating, the OPsensors.deck.gps001 are not updating. Not sure that has anything to do with anything, but thought it worth mentioning. 

[image]

current deltas:

> {
> 	"updates": [{
> 		"$source": "deck.gps001",
> 		"values": [{
> 			"path": "navigation.position.altitude",
> 			"value": 46.0000000
> 		}]
> 	}]
> }

With timestamp:

> {
> 	"updates": [{
> 		"$source": "deck.gps001",
> 		"timestamp": "2017-05-03T09:13:00Z",
> 		"values": [{
> 			"path": "navigation.courseOverGroundTrue",
> 			"value": 4.1008257
> 		}]
> 	}]
> }

sailoog, 2017-05-03:
I am afraid timestamp format is wrong on documentation or signalk-server-node is not using the standard. 

It should be: 

YYYY-MM-DDTHH:MM:SS.SSSZ =  "2017-05-03T15:51:09.331Z"

and not:

YYYY-MM-DDTHH:MM:SSZ =  "2017-05-03T15:51:09Z"

If you do not include timestamp in deltas, the server will do it for you. Please confirm this is working for you and let me suggest a change in signal k documentation for this and an addition for $source/source.

toqduj, 2017-05-03:
Output without timestamp is as described in the previous post, and the screenshot. I've now modified the timestamp to reflect what you wrote, and now the signalk diagnostic input correctly updates. The instrument panel shows no values for the OPsensors.deck.gps001 source, but only for the no_source source... The node-red only shows some values being updated, not all five. No information is passed on to OpenCPN. 

I think the signalk spec timestamp is based on ISO 8601. That standard allows any value to be dropped off the least significant end. This would mean the value without the decimal seconds may be valid too, I think.

sailoog, 2017-05-04:
Ok, I will fix signal k diagnostic to digest both formats.

That no_source source makes me think that you do not have the latest openplotter beta. Are you running 0.11.3?

OpenCPN does not understand signal k data. If this is your goal, maybe delta messages is not a good idea and it would be better to send NMEA 0183 to kplex and these data will be passed to opencpn and translated to signal k.

toqduj, 2017-05-04:
I have 0.11.3 alpha according to the "about". 

For now, Delta messages will do. I'm not so interested in getting OpenCPN to work with this, instead I want to further develop the SK sending sensors concept.

toqduj, 2017-05-05:
As for the timestamps, since 8601 allows anything to be dropped off the end, perhaps it's wise to initialise the timestamp in OP as "2017-05-03T15:51:09.331Z" (current date and time), and replace from [0] with whatever timestamp comes in from the SK message. That way, when the incoming timestamp is "2017-06-02T12:12Z", it would replace the first X characters from the initial timestamp to:  "2017-06-02T12:12:09.331Z". Something like:

`
timestamp = u"2017-05-03T15:51:09.331Z" # or generated from the system time
ts[0 : (len(incomingTimestamp) - 1)] = incomingTimestamp
`

toqduj, 2017-05-07:
I get slightly better display using the following Node-Red, previously, lat and lon were not showing any value. Although I now noticed a small bug: CoG must be in degrees, not radian (I got confused by the SK documentation saying they stuck to SI for everything except position). Easy fix though. 

Too bad there isn't a node-red XY plotter that can take both latitude as well as longitude :). 

[image]

toqduj, 2017-05-07:
The environmental sensors are now also sending updates on temperature, humidity, pressure, and illuminance. No timestamp, so the messages look like this:

`{
	"updates": [{
		"$source": "deck.env001",
		"values": [{
			"path": "environment.outside.humidity",
			"value": 36.7169799
		}]
	}]
}`

toqduj, 2017-07-24:
Everything seems to work fine now.

closed 2017-07-24 by toqduj