sailoog/openplotter #158: Error when starting SDR

syska, 2017-05-15:
I get this error when starting SDR from the command line (the shortcut does not work)

I'm not sure how much more information you need. 

I have no expirience with python.

Here is what I get:
pi@openplotter:~/.config/openplotter/tools $ python SDR_AIS.py 
Traceback (most recent call last):
  File "SDR_AIS.py", line 235, in <module>
    MyFrame().Show()
  File "SDR_AIS.py", line 111, in __init__
    self.button_vhf_Tx.Disable()
AttributeError: 'MyFrame' object has no attribute 'button_vhf_Tx'

Python is version 2.7.9

sailoog, 2017-05-15:
What is your openplotter version?

syska, 2017-05-15:
0.9.0alpha ...

I guess I should try to update ... Let me see if I can figure out how to do that 😀

syska, 2017-05-15:
So this fixed the issue. Next issue.

closed 2017-05-15 by syska