sailoog/openplotter #159: Can't uncheck "Enable access point"

syska, 2017-05-15:
This is a weird issue I guess ... or else just not sure what the heck is going on.

"Access point device" dropdown list is empty.
In the "Sharing internet device" i have "wlan0", "eth0" and "none".

If I do i "ifconfig -a" I also get "wlan0", "eth0" and "lo".

So ... what am I to expect in the "Access point device" dropdown? I would say "wlan0" or something like that ... or is my wifi module that's not working anymore? ( upgrade from version 0.9.0alpha )

I might be getting a "raspberrypi 3" later this week ... just to avoid using a USB port on the wifi module.

Please let me know if you need more information.

sailoog, 2017-05-16:
Not much sense...
Some wifi dongles stop working due to kernel updates, do you know your dongle chipset?

syska, 2017-05-17:
@sailoog 

No, really not much sense of this ... then I bought a RaspberryPi 3 👍 and it started to work. (The other was a RaspberryPi 2)

Did a clean install of 0.10.0 ... nothing more than that. It's working now ... 

The dongle i used was this one: http://www.tp-link.dk/products/details/cat-11_TL-WN725N.html

I also have a WiPi dongle ... same issue.

Both dongles works on my Windows 10 machine. Both dongles works on Windows 10 IOT on the Rasp.

Not sure if it's worth investigating more if no one else have the same issues. But now it's here if anyone else expirence the same I did ... was kind of in a deadlock not being able to disable the AP on the rasp.

closed 2017-05-18 by syska