sailoog/openplotter #165: Don't know where to start

benjamingotke, 2017-06-19:
Hi.
I have a Raspberry PI3 running the newest openPlotter software. I have a NMEA2000 network, for now only wind and a gps sensor connected with an Actisense NGT-1. I had problems with getting the data to OpenCPN, but that is solved now.
I allso have a senseHat to my Raspberry, and I would like SignalK to take temperature and pressure and humiditi to N2K, so I can see it in OpenCPN, Node-red and on my Garmin Plotter onboard.
How do I do that?
I'm not familiar with codes yet, but my feeling is that it will go fast....

I allso have a swichboard from Denkovi. This one: http://denkovi.com/internet-ethernet-12-channel-relay-board-daenetip4-io-snmp-web

Is it possible to control this by NodeRED? and how to do that?

Last questions; I can easely download charts to OpenCPN with the carts downloader, but I have difficult finding charts over caribbean, South Pacific and Australia. Anyone who knows where I can get them?

/Benjamin

toqduj, 2017-06-25:
last question: does openseamap has anything in the areas you're looking for?

benjamingotke, 2017-06-26:
@toqduj ; I finally found a version of CM93. That works perfectly. World covering :)

benjamingotke, 2017-06-26:
If anybody is good at Nodered and Freeboard, maybe they can help by VNC?

sailoog, 2017-06-28:
Maybe it would be more effective to use http://forum.openmarine.net to these general and How to questions.

Quick reply...

You should be able to get data from your sensehat using I2C tab, then you can send signal k data to N2K network using Tools -> NMEA 200 generator and allowing input for those PGNs on your Actisense NGT-1

Caribbean, South Pacific and Australia charts are not free. You may use S-63 charts but they are not cheap.

benjamingotke, 2017-06-28:
Hi
I manage to get openCPN to work. 

I have opened several questions on other foras which is more sutable, i hope. I'm sorry to ask all these questions, but I love this and I hate when I'm stopped because I can't make it work. 

I'll try the NMEA2000 generator in a moment :)

benjamingotke, 2017-06-28:
Right now im working on changing eth0 to static thue this: http://forum.openmarine.net/showthread.php?tid=599
It does not matter what I do, then OpenPlotter overwhrite.. 
I've tried this thread too: http://forum.openmarine.net/showthread.php?tid=22&highlight=dhcp
and a thread linked to this: https://github.com/sailoog/openplotter/blob/master/wifi_server.py
But i dont know how to find that file...

benjamingotke, 2017-06-28:
@sailoog 
I just tried to put it thrue nodered and SignalK
I get a lot of debugs like this;
28/06/2017, 22:55:43node: 88e00f61.23759
msg.payload : Object
object
environment.inside.humidity: 31.0938892365
28/06/2017, 22:55:44node: 88e00f61.23759
msg.payload : Object
{ environment.outside.pressure: 102016.015625, environment.inside.temperature: 310.183332825 }
28/06/2017, 22:55:44node: 88e00f61.23759
msg.payload : Object
{ environment.inside.humidity: 31.4992923737 }
28/06/2017, 22:55:45node: 88e00f61.23759
msg.payload : Object
{ environment.outside.pressure: 102016.162109, environment.inside.temperature: 310.179167175 }

And nothing is showend in the dashboard..

My flow looks like this:

`[{"id":"217cab7.d842f54","type":"tab","label":"Flow 1"},{"id":"88e00f61.23759","type":"debug","z":"217cab7.d842f54","name":"","active":false,"console":"false","complete":"payload","x":144,"y":126,"wires":[]},{"id":"1c8b3d5e.821e43","type":"websocket in","z":"217cab7.d842f54","name":"","server":"bf7d28f7.f45188","client":"","x":194,"y":46,"wires":[["ee667c0b.37b94"]]},{"id":"b403753d.727228","type":"ui_gauge","z":"217cab7.d842f54","name":"courseOverGroundTrue","group":"9964cc9.930fa3","order":1,"width":0,"height":0,"gtype":"compass","title":"Course Over Ground","label":"","format":"{{value | number:1}}","min":0,"max":"360","colors":["#00b500","#e6e600","#ca3838"],"seg1":"","seg2":"","x":664,"y":306,"wires":[]},{"id":"96a571fd.36da9","type":"ui_text","z":"217cab7.d842f54","group":"9964cc9.930fa3","order":3,"width":0,"height":0,"name":"longitude","label":"Longitude","format":"{{msg.payload | number:8}} &deg;","layout":"row-spread","x":614,"y":426,"wires":[]},{"id":"ee667c0b.37b94","type":"jsonpath","z":"217cab7.d842f54","expression":"$.updates..values[*]","split":false,"name":"Signal K delta values","x":494,"y":46,"wires":[["c8bc6219.653cb"]]},{"id":"c198eef1.7f80c","type":"freeboard","z":"217cab7.d842f54","name":"node-red-freeboard","x":174,"y":186,"wires":[]},{"id":"c8bc6219.653cb","type":"function","z":"217cab7.d842f54","name":"merge values","func":"var paths={};\nvar arrayLength = msg.payload.length;\nfor (var i = 0; i < arrayLength; i++) {\n    paths[msg.payload[i].path]=msg.payload[i].value;\n}\nmsg.payload = paths;\nreturn msg;","outputs":1,"noerr":0,"x":714,"y":46,"wires":[["c198eef1.7f80c","bc5d23b4.ecb63","88e00f61.23759","ba1e0210.964e8","ed9c497a.e56588","7cad1a28.b5bf34","7f45010c.82eb","33f4fae.104eb06","82598552.dda128"]]},{"id":"de3743b8.48de1","type":"ui_gauge","z":"217cab7.d842f54","name":"speedOverGround","group":"9964cc9.930fa3","order":2,"width":0,"height":0,"gtype":"gage","title":"Speed Over Ground","label":"kn","format":"{{value | number:1}}","min":0,"max":10,"colors":["#00b500","#e6e600","#ca3838"],"seg1":"","seg2":"","x":644,"y":246,"wires":[]},{"id":"bc5d23b4.ecb63","type":"function","z":"217cab7.d842f54","name":"navigation.speedOverGround","func":"var signalk_key=\"navigation.speedOverGround\";\n\nif(msg.payload.hasOwnProperty(signalk_key)){\n    msg.payload=msg.payload[signalk_key];\n    return msg;\n}","outputs":1,"noerr":0,"x":204,"y":246,"wires":[["7c0214b9.5d94dc"]]},{"id":"7c0214b9.5d94dc","type":"function","z":"217cab7.d842f54","name":"m/s to knots","func":"msg.payload = Number(msg.payload)*1.94384\nreturn msg;","outputs":1,"noerr":0,"x":444,"y":246,"wires":[["de3743b8.48de1"]]},{"id":"ba1e0210.964e8","type":"function","z":"217cab7.d842f54","name":"navigation.position.latitude","func":"var signalk_key=\"navigation.position\";\n\nif(msg.payload.hasOwnProperty(signalk_key)){\n    msg.payload=msg.payload[signalk_key][\"latitude\"];\n    return msg;\n}","outputs":1,"noerr":0,"x":194,"y":366,"wires":[["d28e4637.195638"]]},{"id":"d28e4637.195638","type":"ui_text","z":"217cab7.d842f54","group":"9964cc9.930fa3","order":4,"width":0,"height":0,"name":"latitude","label":"Latitude","format":"{{msg.payload | number:8}} &deg","layout":"row-spread","x":614,"y":366,"wires":[]},{"id":"7cad1a28.b5bf34","type":"function","z":"217cab7.d842f54","name":"navigation.position.longitude","func":"var signalk_key=\"navigation.position\";\n\nif(msg.payload.hasOwnProperty(signalk_key)){\n    msg.payload=msg.payload[signalk_key][\"longitude\"];\n    return msg;\n}","outputs":1,"noerr":0,"x":194,"y":426,"wires":[["96a571fd.36da9"]]},{"id":"ed9c497a.e56588","type":"function","z":"217cab7.d842f54","name":"navigation.courseOverGroundTrue","func":"var signalk_key=\"navigation.courseOverGroundTrue\";\n\nif(msg.payload.hasOwnProperty(signalk_key)){\n    msg.payload=msg.payload[signalk_key];\n    return msg;\n}","outputs":1,"noerr":0,"x":214,"y":306,"wires":[["b403753d.727228"]]},{"id":"7f45010c.82eb","type":"function","z":"217cab7.d842f54","name":"environment.inside.temperature","func":"var signalk_key=\"environment.inside.temperature\";\n\nif(msg.payload.hasOwnProperty(signalk_key)){\n    msg.payload=msg.payload[signalk_key];\n    return msg;\n}","outputs":1,"noerr":0,"x":224,"y":493,"wires":[["eeabbe77.d0505"]]},{"id":"33f4fae.104eb06","type":"function","z":"217cab7.d842f54","name":"environment.outside.pressure","func":"var signalk_key=\"environment.outside.pressure\";\n\nif(msg.payload.hasOwnProperty(signalk_key)){\n    msg.payload=msg.payload[signalk_key];\n    return msg;\n}","outputs":1,"noerr":0,"x":229,"y":554,"wires":[["55b4865d.e6c098"]]},{"id":"82598552.dda128","type":"function","z":"217cab7.d842f54","name":"environment.inside.humidity","func":"var signalk_key=\"environment.inside.humidity\";\n\nif(msg.payloa [... cut ...] "9964cc9.930fa3","order":0,"width":0,"height":0,"name":"Humidity","label":"Humidity","format":"{{value | number:3}}","layout":"row-spread","x":569,"y":602,"wires":[]},{"id":"bf7d28f7.f45188","type":"websocket-listener","z":"","path":"ws://localhost:3000/signalk/v1/stream","wholemsg":"false"},{"id":"9964cc9.930fa3","type":"ui_group","z":"","name":"Default","tab":"bdb776e4.17a1e8","order":1,"disp":false,"width":"6"},{"id":"bdb776e4.17a1e8","type":"ui_tab","z":"","name":"Home","icon":"dashboard"}]`

closed 2017-08-27 by sailoog