sailoog/openplotter #170: Requesting deltas from OP SignalK server via MQTT for display.

toqduj, 2017-07-24:
Hello Sailoog, 

With the WiFi sensors now working satisfactorily, the next project involves making a WiFi display unit. I intend to use similar components again: an MKR1000, a 2.9 inch (wide) e-paper display, and a Skywriter for input/selection. This match would enable me to put everything in a waterproof box with a minimum of holes for external connections (only one for power). 

The question is, how can I request values from the Openplotter server? The MQTT tab only offers options for Signal K input, but not output. Also, if the display is only displaying speed (for example), maybe it doesn't make sense to send the entire SK schema all the time, so the display should be able to:
  1) find out the available values in the SK structure that are being updated (full SK schema download, but only with actively updated values). This would work for low update frequencies.
  2) maybe subscribe to a SK (sub) topic to receive the deltas (for everything or only a particular (sub) structure)

I tried to find out if something like this is already available, but couldn't see any. Is such a mechanism already in existence? If not, do you have a suggestion on what would be the easiest to implement for you?

Cheers, 

B.

richardmbell, 2017-07-25:
Perhaps I am missing something, but if you are publishing to an mqtt server why can't you subscribe two whichever topic you want?  I have never done that on an Arduino. Is that possible?

toqduj, 2017-07-25:
That would only work for data coming in from the MQTT-enabled sensors, and only if you know which topics are available and what is being published to a particular topic. It would not work for server-generated derivatives or non-MQTT sensors. I'm therefore wondering whether there is a general method in place to find and read values in the SK tree via MQTT. 

Subscribing and ingesting JSON on the Arduino is not a problem, but you'd need an arduino with sufficient memory to store the JSON messages. I'm currently playing with the ARM Cortex-M0 based boards for that.

toqduj, 2017-08-07:
@tkurki mentioned there are MQTT topics for each parameter, so I'll be looking into that..

toqduj, 2017-08-13:
Don't seem to be working on OP. @sailoog any chance that we'd be able to create output topics in the interface in the same way that we can create input topics? I.e. in the "Add MQTT topic" dialog, to have under "Type" the option to say "Signal K key output", "Signal K delta output", and "Signal K full output"?

tkurki, 2017-08-13:
I would like to have a standardised mapping between SK and MQTT, like
- Delta queue is `signalk/delta`, where data is JSON in standard SK delta format
- invividual values are available for read/write at topics named like `signalk/values/vessels/your-vessel-signalk-id-here/navigation/courseOverGround`, where the data would be floats as strings

What did you mean with "Signal K key output" and "Signal K full output"?

toqduj, 2017-08-14:
OP has a user interface to set up the MQTT topics. The topic can be set freely, the content (input only at this point), can be selected to be: 
"Signal K key input": a topic whose input is supposed to be just single values. These get mapped to a particular SK key. 
"Signal K delta input": A topic whose input can be Signal K deltas. This is what I use. 

I was hoping to be able to get:
 - a key output topic, spitting out single values for a particular key when they get updated, like the signalk/values/vessels/your-vessel-signalk-id-here/navigation/courseOverGround topic you suggested, or
 - a full or delta output topic, from which I can glean which keys are updated and available within vessels/self.

tkurki, 2017-08-14:
mqtt-gw plugin provides what you're after I think. Take it for a ride. There isn't much code, easy to see what it is doing.

toqduj, 2017-08-15:
Ok, I'll have to try and install it on top of an OP install. Any tips for that?

tkurki, 2017-08-15:
How fresh is the OP Signal K server and how do you update it? Other than that the mqtt gw is available in the SK Node server app store.

But if I were you I would install a local version of SK server on whatever desktop machine I use and play around with that.

toqduj, 2017-08-16:
The OP server is the very latest beta, updated via the OP interface. Isn't that a SK-node install anyway? It's tempting to try to install anew with https://github.com/SignalK/signalk-server-node , but I suspect I'll be losing functionality I'd otherwise have automatically, requiring some manual labour to get this back up again. 

Then again, the node "shop" is nice...

toqduj, 2017-09-02:
I have managed to install the mqtt-gw plugin from the SK node server app store on top of the openplotter install. It wasn't completely hassle-free, I had to install the following packages before the gw install worked:

(error, missing zmq.h)
>>> sudo apt-get install libzmq3-dbg libzmq3-dev libzmq3

(next error, missing dns-sd.h)
>>> sudo apt-get install libavahi-compat-libdnssd-dev

After this, it installed. I haven't looked at whether it outputs data yet.

toqduj, 2017-09-03:
Bonus: after installing libnss-mdns and avahi-utils, it now advertises the SK services over Bonjour. Useful for e.g. WilhelmSK to autodiscover the "boat"

tocan, 2017-09-04:
I am trying at the moment to conect my EPEVER Tracer with signal k. Can someone help me in this topic? I like to use influx db,  and pymodbus library.  I do not use cable I do complete all with an ESP8266 as a virtual hub with so cat.

tkurki, 2017-09-04:
@tocan I use ESP8266 with my SK server for battery monitoring (voltage & current) by sending Signal K deltas to SK server over UDP. The server then stores all SK data in InfluxDb, with Grafana for graphs.

However this is not the correct issue for getting support for this. Please join signalk-dev.slack.com or use OpenPlotter forum at http://forum.openmarine.net/forumdisplay.php?fid=1.

tocan, 2017-09-04:
for signalk-dev.slack.com i need a invitation and in the forums of open
marine there seems not so much interest to develope together a solution but
i see in downloads that a lot of people seem to have epever A or B series
at their boat. I am not a programmer and englih skills are rare.  We do
here all things with try and error and for signal k and such things you
need advanced knowledge because still the documentation sucks in step by
step examples. If i get an intrance i can do like i have done in steps a
documentation of signal k in German language and than romanian. but the use
of git we do not understand so a translation we can do at sourceforge is
more easy to handle for us.

regards Daniela

dmdelorme, 2018-02-14:
Hi Tocan,
did you find my example of sending signalK from a Tracer charge controller. The Git is a mess but i have been running the code continuously for a while now. I run it via a sh script every 5 seconds. I run node-red on on my laptop and point it to the SK server, or access the SK-instruments page. I ordered the Wifi adapter for the tracer but due to a change in the WiFi chip it was useless.

sailoog, 2019-03-20:
MQTT has been rebuilt and integrated into actions. closing.

closed 2019-03-20 by sailoog