sailoog/openplotter #171: Automatic update of grib files

mirthestam, 2017-07-25:
I would like to propose a new feature.

Automatic downloads, (daily, or hourly) of updated grib files.
This would be nice for OpenCPN VNC users. This way OpenCPN always has the latest available data and there is no need to manually download them using zyGrib.

A new tab page could be used for this. Available properties could be the lat/lon min/max as well as a checkboxlist containing wind, temperature, etc, and a interval combo (hourly, daily, etc).

pcav, 2017-07-25:
Agreed, a very useful improvement. Will this be upstream in OpenCPN?
Thanks.

mirthestam, 2017-07-25:
OpenCPN grib plugin automatically selects the latest file when opening the plugin, so it supports this out of the box.

mirthestam, 2017-07-27:
Apparently the zyGrib server does not allow direct access. Therefore, implementing this could get users banned. Is anyone aware of other grib servers we could use?

sailoog, 2017-07-28:
I like it. I vote yes. 
We could use more than one source, this is the easier one: http://www.globalmarinenet.com/free-grib-file-downloads/#download

mirthestam, 2017-07-29:
I got a proof of concept 'download' working with the NOAA GFS 0.25 Degree data set. I'll look into scheduling as well as a UI next. I'll push to git to share after the PoC is done so we can shoot and collaborate.

sailoog, 2017-07-29:
Nice! please push to beta branch https://github.com/sailoog/openplotter/tree/beta

mirthestam, 2017-07-30:
I've added a pull request to further discuss the functionality.

mirthestam, 2017-08-31:
A few conclusions as discussed in pull request #172 

First  we implement this as a developer built tool located `~/.config/openplotter/tools` based upon https://sailoog.gitbooks.io/openplotter-documentation/content/en/standalone-tools.html.  This way we can continue development in 'Beta' and disable the tool by default.

In the future we integrate it into OP. Integration will include a new 'Weather' tab containing:
- A manual update button, including a progress popup. 
- Input fields for the Lat, lon boundaries as well as the number of days
- Checkboxes for the required fields (i.e. lon, lat, cape).

Automatic triggers will then be delegated to the OP action/trigger system. We'll add a temporary time interval property until the addition to the action/trigger system has been implemented.

sailoog, 2017-08-31:
Thanks @mirthestam 
just some tips to make easier starting with tools.

If you want to access to OpenPlotter classes from a core tool you should start wit this:

```python
import os, sys
op_folder = os.path.join(os.path.dirname(os.path.realpath(__file__)), '..')
sys.path.append(op_folder+'/classes')
from language import Language
```
and to create a config file in ~/.openplotter/tools for that tool:

```python
from classes.conf import Conf2
conf_file = 'my_tool.conf'
conf2 = Conf2(conf_file)
```
and finally, to open the config file when "settings" button is pressed:

```python
import subprocess
home = os.path.expanduser('~')
if 'root' in home:
	home = '/home/'+os.path.expanduser(os.environ["SUDO_USER"])
if len(sys.argv)>1:
	if sys.argv[1]=='settings':
		subprocess.Popen(['leafpad',home+'/.openplotter/'+conf_file])
	exit()
```

toqduj, 2017-09-07:
Speaking of Gribbing stuff, This lot seems to have a nice weather forecast for the med:
http://openskiron.org/en/

Cheers, 

B.

mirthestam, 2017-12-04:
I'll pickup on this soon. So that we can have it ready by next season :)

sailoog, 2019-03-20:
Abandoned feature. closing. reopen if necessary.

closed 2019-03-20 by sailoog