sailoog/openplotter #175: Add a logger to log navigation data

aayaffe, 2017-08-05:
I am suggesting to add an option to log NMEA messaged to files. 
This log can later be used to analyze a certain sail or for debugging purposes.
The NMEA 0183 tab should inlclude the option to add a input\output from file.
The output option should include a flag for maximum size of data file.
It seems that KPLEX has the option to read and write from a file

toqduj, 2017-08-06:
Why not just log the SK data? 0183 may contain only a subset of the data.

sailoog, 2017-08-06:
Editing kplex.conf you can log NMEA 0183 into a file in advanced mode. But logging SK would be great.
Do you know if there is any logger/player SK plugin planed?

tkurki, 2017-08-07:
You can add a `log` element to a provider pipeline and depending on where in the pipeline you add it it will log raw input data or SK data that is produced - or you can log both by adding two log elements.

https://github.com/tkurki/cassiopeia-settings/blob/master/signalk-server-settings.json#L18-L22

The log is semicolon separated rows, timestamp - discriminator - data. Discriminator is used to tell what format the row is in: A is "Actisense serial" (canboat fast format N2K), N is NMEA0183, I is Signal K.

tkurki, 2017-08-07:
Related: https://github.com/SignalK/signalk-server-node/pull/264

sailoog, 2017-08-07:
Yes! that is what we need, thanks @tkurki

tkurki, 2017-08-07:
FYI https://github.com/SignalK/signalk-server-node/blob/master/providers/multiplexedlog.js can play back these files.

sailoog, 2017-08-07:
Great. I guess the server should be restarted with just that provider and restarted again with normal providers. Any example?

tkurki, 2017-08-10:
`bin/signalk-server -s settings/multiplexed.json --filename ../../../Downloads/signalk-rawdata.log.2017-07-27T09`

dmdelorme, 2017-11-19:
A little late to comment,  but it would be a nice feature. I save all my trips but i would like to collect the skdata, When i am under way or anchored. I will have to check my gpx files to see what is included as i currently safe all my trips during the during the season when the water is not solid "summer" . I would like a easy button or log skdata automatically when something is happening ie heel, pitch, MOB, speed, ...
I sail solo so it would be nice if i had a trigger that would kill the auto pilot or trigger the boat to heave-too.  possibly a nfc tag on a life vest.  like the clip on my out board that kills the engine when the clip is pulled.
I have gone though this scenario a many times if some one falls over board how do you automatically log the event.    the MOB alarm will not be used unless a big red button is at hand and i am not alone highly unlikely. The chance of this happening is small since I  take precautions but  consequence is large. I mitigate the issue with properly installed jack lines etc. but it would be nice if my boat stopped or came back for me. 
as it currently stands if i get on the water side of the boat with auto pilot my boat will continue on. I have to deal with the water and re boarding and i have to deal with the force of the water.  I have swift water rescue training and i know i could not climb a rope with a 5 kn speed. . so grabbing a tailing line and pulling is not a reasonable option.
One of many is winter projects is a re-boarding device in my PFD  A rope ascender i can clip on to my life line when i on the wrong side of the boat and someone to test it with me when it is warm.

skipperearly, 2017-12-01:
@dmdelorme: (maybe also late ;) - wouldn't a solution to your issue raised be to use a personal AIS flare ( like https://seaangel.at/index.php?language=en) and add a list with this mmsi in OP action options which, when received disable the autopilot.  
Have also seen discussions on taking action when the bluetooth connection is broken with the mobile phone. 
Maybe both options could be added to the action list in OP

JeroenAdam, 2018-04-11:
Very nice proof of concept for datalogging using Node Red and SQLite on Raspberry pi
https://flows.nodered.org/flow/36f2de38ed4bb62cf3a5e8fdc76d779e
https://youtu.be/ccKspiI8FRw

tkurki, 2018-04-12:
SK server now has logging that is configurable from the admin ui: activation per provider and log file location. Then there is the signalk-to-influxdb plugin to write all numeric data and positions into InfluxDb.

tkurki, 2018-04-12:
...and you can configure the server to play back a log file from the admin ui as well, no settings file editing needed.

sailoog, 2018-04-16:
New OP version will go SK server way to manage logs.
On first stage we will use files and on second stage influxdb when grafana is added too

tkurki, 2018-04-16:
Just to be clear: i think raw data logging is separate from InfluxDB, you achieve different things with them.

sailoog, 2018-04-16:
ops, so can not you replay logged data from InfluxDB? sorry not tested signalk-to-influxdb yet.
To be clear too: You replay logged data adding a file provider and disabling rest of providers, don't you?

tkurki, 2018-04-16:
That is correct

dmdelorme, 2018-04-22:
@JeroenAdam I have moved away from Node-red for Mission critical stuff i find buggy and resource intensive. I do use it to display SKdata but run node red my laptop.
I have been playing with a arduino nano NRF24L01+ RPI master Very low power send and receive.
Currently reporting wind Direction and Speed to RPI that writes to SKserver on a Openplotter RPI.  it is powered by 12 V to mast head.
Anchor light is controlled by Mosfet via nano. command to turn off and on via RPI.
 I used the nano because it is 12V tolerant. The anchor light and Wind
instruments  needs 12V.

> Sent the message: ['G', '0']
Received: [32, 49, 46, 54, 44, 51, 49, 51, 0, 0]
Our received message decodes to:  1.6,313
1.6, 5.46288
{"updates": [{"source": {"type": "wind","src" : "arduino"},"values":[{"path": "environment.wind.speedApparent","value":1.6},{"path": "environment.wind.angleApparent","value":5.46288}]}]}

Is this a proper SK statement:
`{"updates": [{"source": {"type": "wind","src" : "arduino"},"values":[{"path": "environment.wind.speedApparent","value":1.6},{"path": "environment.wind.angleApparent","value":5.46288}]}]}`

ardunio serial output

>Turning off the radio.
G0
Looks like they want a string!
G0
sensor Value 80.00
Wind speed: 0.00
Wind dir: 68.00
0
 0.0, 68
We sent our message.

A live jacket version could run on a battery. 
Tether if connection lost or other message. start event.
// Performance using  InfluxDB
Is any one running one with openplotter??
how big of a database is it?
SSD?
 When i sail i like it stupid and easy, like me . I use openCPN for logs and http://openplotter.dmdelorme.ca:3000/@signalk/instrumentpanel
for additional Instruments.

A Water balloon canon/ pirate flag controller hat might be fun. EH!
[image]

.

sailoog, 2019-03-20:
OP v2.x.x will use SK logs feature. closing.

closed 2019-03-20 by sailoog