sailoog/openplotter #176: OpenPlotter 14.4 UnboundLocalError: local variable 'magn_list' referenced before assignment

pascalpci, 2017-08-09:
Just updated and impossible to run openplotter
when i comment line 1873 to 1875 /home/pi/.config/openplotter/openplotter", openplotter runs
maybe a problem from openplotter.conf migration ?

thanks you for this project, i follow it often

dmdelorme, 2017-08-09:
I have the same problem ditched my old conf. I am also noticing hi processor usage from Python.

pascalpci, 2017-08-09:
@dmdelorme 
You are rigth thank you
For me just delete all I2C section in  openplotter.conf and it works
I am also noticing 99% CPU for python with top command

sailoog, 2017-08-09:
Yes, we have changed the IMU management and that error means that you had an IMU defined on I2C tab. 

To fix this edit /home/pi/.config/openplotter/openplotter.conf and delete all I2C defined sensors:

```
[I2C]
sensors =
```

Now, this is how IMUs work: https://sailoog.gitbooks.io/openplotter-documentation/content/en/compass.html

Keep this issue open to help others please.

sailoog, 2017-08-09:
The processing increase is due to the new IMU management, we will try to optimize that.

dmdelorme, 2017-08-09:
thanks sensors and compass are working much better no lag. I bought a tanzer 26 this spring it is my first sailboat i have it set up and it works fine. the calibration is very slick. still need to find a good sunlight display. and a case to hold it

pascalpci, 2017-08-09:
Thanks a lot !
If you need a translator in French, I can help you

sailoog, 2017-08-09:
@pascalpci yes, we will need to translate the interface soon, we are reaching the beta stage. I will notify when we can start translating interface.

And we need also French translators for documentation, here we try to coordinate translations: http://forum.openmarine.net/showthread.php?tid=507

closed 2017-08-29 by sailoog

sailoog, 2017-09-02:
@pascalpci we have started interface translations of beta version: https://crwd.in/openplotter