sailoog/openplotter #178: NOOBS don't see the OS

border0464111, 2017-08-14:
Hi I tried to download the zipfile from github and install it on my SD card, the NOOBS install menu open but there is no OS to install. 

I downloaded and uploaded on my SD card the last version of NOOBS and add openplotter into the OS folder. When I boot my raspberry, I see others OS but not openplotter. 

Note that I have a raspberryB (the second generation )  

What Can I do, to investigate futher ?

toqduj, 2017-08-16:
If you want to install the dev version, start from this image, and, once installed, update the OS and update OP. Both can be done in the openplotter interface. 
http://sailoog.com/en/blog/download-openplotter-rpi-v0100-noobs

sailoog, 2017-08-21:
OpenPlotter NOOBS is not raspberry 1 compatible, only raspberry 2 and 3.

closed 2017-08-21 by sailoog