sailoog/openplotter #180: Include Avahi for Signal K discovery

tkurki, 2017-09-03:
Apparently the current setup does not include Avahi (os level support) and Mdns (Signal K server level support). They are needed for Signal K client apps to autodiscover the server.

https://github.com/sailoog/openplotter/issues/170#issuecomment-326791835

sailoog, 2017-09-04:
OK. I remember that being on the to do list, what I can not remember is why this jumped. I will check if there is any incompatibility with the current Wifi AP management. Thanks.

@tkurki can I have your email? I have some questions about SK server. Mine is info[youknowwhat]sailoog.com

toqduj, 2017-09-13:
On my test system running the latest OP beta, it shows no obvious problem. Both with bridged eth0 and without, the AP section and functionality works fine. With avahi and mdns installed.

sailoog, 2017-09-15:
I will add this to the next major update. Did it need any extra configuration or just installing?

@tkurki we are close to the stable version and waiting for this to make a major update: https://github.com/SignalK/specification/issues/244 any news?

tkurki, 2017-09-15:
No configuration needed, Node server just checks if npm module `mdns` is available, which in turn needs avahi.

tkurki, 2017-09-15:
As for enums let's continue discussion at https://github.com/SignalK/specification/pull/382.

sailoog, 2017-10-07:
FYI

node v6.11.4
signalk-server-node 0.1.32

*** WARNING *** The program 'node' uses the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node>
*** WARNING *** The program 'node' called 'DNSServiceRegister()' which is not supported (or only supported partially) in the Apple Bonjour compatibility layer of Avahi.
*** WARNING *** Please fix your application to use the native API of Avahi!
*** WARNING *** For more information see <http://0pointer.de/avahi-compat?s=libdns_sd&e=node&f=DNSServiceRegister>

tkurki, 2017-10-07:
This is normal.

tkurki, 2017-10-07:
Do you install SK server from npm?

sailoog, 2017-10-07:
Yes

tkurki, 2017-10-07:
All right, I’ll pay more attention to publishing there. So far I have assumed that most people install from Github, that’s why I asked.

sailoog, 2017-10-08:
ops sorry I misunderstood the question. I use github.

closed 2017-10-11 by sailoog