sailoog/openplotter #183: Incoming MQTT SK Deltas not processed in 0.16

toqduj, 2017-10-11:
I've reinstalled OP from a fresh image, and upgraded it to 0.16. However, there appears to be a problem with the processing of the incoming SK messages. 

The sensor connection to the OP Wifi access point runs fine, and it subscribes to the right MQTT topic. Messages are published and received on this topic, which I can check by setting the topic type in the OP configuration window to "General" instead of "Signal K delta input". The SK diagnostic window then shows the incoming SK deltas (see screenshot). Switching the type to "Signal K delta input", however, does not lead to the keys and values as it did with 0.15. 

I don't think I made a stupid mistake, but I'm always open to being corrected on this :). 

[image]

sailoog, 2017-10-11:
Please open openplotter from terminal, set delta type and press "restart". Any error should be printed in the terminal. If not, go to Signal k tab and press restart.

toqduj, 2017-10-14:
Alright, I get a couple of issues scrolling over, it's probably only the last one which is relevant. In order of appearance: 
  - No settings/defaults.json availalble
  - A warning from Avahi that node is using the Apple Bonjour compatibility layer instead of the native Avahi API 
  - error reading ~/.config/signalk-server-node/public/mapcache
  - and repeated, continual errors: "Are you sure you are using the correct line terminator? Not going to handle lines longer than 2048 characters". 

Given the last error, I'll have another look at the line terminator in my Delta messages. I suspect at the moment all incoming Delta messages are stuck together by MQTT preventing interpretation.

tkurki, 2017-10-14:
The first three are non issues. Should get rid of 1 and 3, self induced.

How is your producer pipeline configured? The error message is from `liner`, which splits character stream to lines. But you are receiving individual deltas already as mqtt messages: you should just parse them, in your mqtt provider or by passing to from_json.

toqduj, 2017-10-28:
@sailoog did you find any time to look into the issue? No rush, but I'd like to use OP during the December mini maker faire to demo SK.

toqduj, 2018-01-06:
The behaviour in 0.17 is different, but it still doesn't work (it does in 0.15). Now, it seems the messages are interpreted individually, and when I configure the MQTT channel on the OP server to "General", I see the following message coming in in the SK diagnostic window:
{
	'updates': [{
		'$source': 'deck.env001',
		'values': [{
			'path': 'environment.outside.illuminance',
			'value': 20.97715
		}]
	}]
}

This does not validate with jsonlint as the single quotes are not correctly interpreted.
Note, however, that the environmental sensor sends that message not with single quotes but with double quotes as it should be. Somewhere along the line, it appears there's a "replace" that replaces the double quotes with single quotes. 

Switching the MQTT channel to single quotes does not result in any SK input in the diagnostics window. Instead, I get some thread-related error after a while (see screenshot, bottom left terminal, above the "unit" error message I get when switching to the "General" channel type)..

[image]

sailoog, 2019-03-20:
MQTT feature has been integrated into actions in OP v2.x.x. closing this issue, please open a new one if you need for v2.x.x

closed 2019-03-20 by sailoog