sailoog/openplotter #187: Openplotter stopped reading IMU

dmdelorme, 2017-11-23:


dmdelorme, 2017-11-23:
For some reason openplotter 0.16 latest update is not getting IMU data.
I checked the log no hint there.

 RTIMULibDrive
Settings file RTIMULib.ini loaded
Using fusion algorithm RTQF
min/max compass calibration not in use
Ellipsoid compass calibration not in use
Accel calibration not in use
LSM9DS1 init complete
^Cmple rate 108: : roll:1.385205, pitch:1.841897, yaw:-55.864073
```

so RTIMULibDrive sees it? 
my other sensors are seen openplotter and make it to SK.
the compass Calibration hangs when i click boat is level.

dmdelorme, 2017-11-25:
this is debug from open plotter when i use the Compass reset button.
`failed to load /home/pi/.pypilot/pypilot.conf
failed to load /home/pi/.pypilot/pypilot.conf
Process Process-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pypilot-0.1-py2.7-linux-armv7l.egg/pypilot/boatimu.py", line 46, in imu_process
    s.MPU9255AccelFsr = 0 # +- 2g
AttributeError: 'RTIMU.Settings' object has no attribute '**MPU9255AccelFsr**'
`
 I have changed to A ada fruit 9DOF. 
I am getting the same results. RTIMULibDrive sees it but there is a issue with rtimulib and pypilot./ pypilot.conf
`{"imu\/alignmentQ":[1.0,0.0,0.0,0.0],"imu\/compass_calibration":[[0,0,0,30,0],[0,0,0,30],[0,0,0,30,1,1]],"imu\/compass_calibration_locked":false,"imu\/gyrobias":false}
`
It almost seems like i got a miss match between pypilot and rtimulib. the RTIMUlib.ini has no reference to MPU9255.

sailoog, 2017-11-25:
This issue and this one: https://github.com/sailoog/openplotter/issues/185 make me think that your are not using the NOOBS openplotter 0.10.0 updated to the latest version.

You have to compile this modified version of RTIMULib and no any other: https://github.com/openplotter/RTIMULib2

dmdelorme, 2017-11-25:
I did the update though openplotter i was running 0.16. But ya I am sure some how i ended up with the wrong RTIMU as it does now create the proper *.ini..

sailoog, 2017-11-25:
Have you done the update from the update item in openplotter menu or manually?
You can delete rtimulib ini files manually and they should be created automatically

dmdelorme, 2017-11-25:
yes i did the update via openplotter.
just recompiled RTIMULib as instructed
Also tried replacing pypilot just to make sure.
Deleted ini files several times.
Same error complaining about in imu_process s.MPU9255AccelFsr = 0 # +- 2g AttributeError: 'RTIMU.Settings' object has no attribute '**MPU9255AccelFsr**' ini file has not settings for this. MPU9255

dmdelorme, 2017-11-25:
I am going to compare Sean's version of RTIMUlib to see if there is a difference their should not be but Stuff happens

sailoog, 2017-11-25:
I do not think so.
Today I have to test a fresh update from 0.10.0 I will let you know if I have any of your issues.

dmdelorme, 2017-11-25:
I will create a new unit to as i have a few pi3 sitting for dev.
Hey i think i should thank you and every one else working on this project. it gives me a some relive from not being on Tar.

dmdelorme, 2017-11-25:
I found it and it was related to #185. I found an old IMULib.so in my cache and found a even older one in a egg.tmp removed both and bingo the proper ini file was created.
Problem Solved.. Thanks

closed 2017-11-25 by dmdelorme