sailoog/openplotter #188: append Source to Label

dmdelorme, 2017-11-26:
I am having a issue dealing with multiple sensors 
i would like append Source to Label.
my source is listed as the label in Diagnostic-sk-input
{"source":{"type":"CHARGER","src":"epsolar","label":"OPserial"}
now if i add
`{"source":{"type":"CHARGER","src":"epsolar2","label":"OPserial"}`
wont they over write each other.
is their away to set it to this
`{"source":{"type":"CHARGER","src":"epsolar2","label":"OPserial.epsolar2"}`
I can see the type, src in SK so that is not an issue, and i think i understand why OPserial is important.
I can change the path to reflect the source this might be the best solution 
`[{"source":{"type":"CHARGER","src":"epsolar1","label":"OPserial"},"values":[{"path":"electrical.epsolar1.batteries.main.voltage","value":12.62}`

tkurki, 2017-11-26:
Could you describe the real life setup that you have and what sensors you have?

dmdelorme, 2017-11-28:
Ok,
I have 2 Tracer solar chargers
1 is 10 amp not connected to a load bus at the moment, the second is a 40 amp unit.
if i both write to path. 

` "/vessels/*/electrical/batteries/RegExp/current": {
  "units": "A",
   "description": "Current flowing out (+ve) or in (-ve) to the device"
`

such a dummy. Just using wrong key:
   `"/vessels/*/electrical/chargers/RegExp/current": {
    "units": "A",
    "description": "Current flowing out (+ve) or in (-ve) to the device"`
or
`"/vessels/*/electrical/solar/RegExp/loadCurrent": {
    "units": "A",
    "description": "Amperage being supplied to load directly connected to controller"
`
Sorry it was a dumb point just too many choices.

closed 2017-11-28 by dmdelorme