sailoog/openplotter #195: Can't open openplotter

vincenthure, 2018-02-20:
I get this message and don't understand why

./openplotter.py
Traceback (most recent call last):
  File "./openplotter.py", line 2906, in <module>
    MainFrame().Show()
  File "./openplotter.py", line 660, in __init__
    self.set_layout_conf()
  File "./openplotter.py", line 808, in set_layout_conf
    with open(home+'/.config/signalk-server-node/settings/openplotter-settings.json') as data_file:
IOError: [Errno 2] No such file or directory: '/home/pi/.config/signalk-server-node/settings/openplotter-settings.json'

vincenthure, 2018-02-21:
When I make a fresh install on RPI it was working well
After one day it was broken. I suppose that the file "openplotter-settings.json" disappear.
who can I regenerate it?

e-sailing, 2018-03-02:
You are using an old version 0.8 I think. There isn't any file openplotter.py today it is simply openplotter

thezerocool, 2019-01-03:
probably the same issue I am having, I posted a seperate issue as my error is different but the cause of my signalk error was an update from a old version, 1.0 im guessing, to 1.2

closed 2019-03-20 by sailoog