sailoog/openplotter #198: add socketCAN support

Dirk--, 2018-03-28:
since wifi_server.py overwrites /etc/network/interfaces , a first step might be to include
source /etc/network/interfaces.d/*
to allow custom can and vcan interface settings

sailoog, 2018-03-30:
good idea. 
Do you have can and vcan interface settings samples?

pin @e-sailing

e-sailing, 2018-03-30:
You can disable overwrite of /etc/network/interfaces by deleting the line '# generated by openplotter' in file /etc/network/interfaces     and then disable wifi ap (nothing will be overwritten anymore until you click enable wifi ap).
This is an undocumented feature (line 325 in wifi_server.py).

Dirk--, 2018-03-30:
samples: yes, work in progress: https://bastelbude.grade.de/mediawiki/index.php?title=OpenPlotter

sailoog, 2018-04-09:
Added by https://github.com/sailoog/openplotter/pull/200

FredericGuilbault, 2019-02-13:
@Dirk-- does the merhe #200 solved the issue for you ? 
Can we close it ?

Dirk--, 2019-02-14:
Sorry, I did not do any testing (recently working with 'pure' signalk only) ...